Atmosphere-NX/Atmosphere #1880: Program ID 524850534f53454d. error desc karnel panic (0xf00)

JrRamos1524, 2022-05-11:
What's the issue you encountered?
I'm on atmos at 1.3.2, I updated fusee.bin ofc. I got this message when I inject fusee-primary.bin:
"A fatal error occured when running Atmosphere.
Program ID: 524850534f53454d
Error Desc: Kernel Panic (0xf00)
Report saved to /atmosphere/fatal_errors/report_0000000001def6bd.bin"

How can the issue be reproduced?
It happened when I wanted to play and my switch restarted. When I wanted to start my switch and injecting the oficial fusse.bin from my computer or phone “Rekado” appears as an error.

Environment?
Atmosphere 1.3.2 with its fusee.bin
Additional context?

JrRamos1524, 2022-05-11:
>
[report_0000000001d6cf9d.zip](https://github.com/Atmosphere-NX/Atmosphere/files/8666542/report_0000000001d6cf9d.zip)

JrRamos1524, 2022-05-11:
[report_0000000001d6cf9d.zip](https://github.com/Atmosphere-NX/Atmosphere/files/8666605/report_0000000001d6cf9d.zip)

SciresM, 2022-05-11:
You have Atmosphere 1.2.5 installed, for some reason. Please re-open with the issue template filled out if the error still occurs using the latest atmosphere.

SciresM, 2022-05-11:
If you have a file at sd:/atmosphere/mesosphere.bin, delete it. The fatal report you posted was generated from an Atmosphere 1.2.5 kernel.

SciresM, 2022-05-11:
Oh, "inject fusee-primary.bin" <-- do not use fusee-primary.bin. It doesn't exist any more. Use fusee.bin.

closed 2022-05-11 by SciresM

JrRamos1524, 2022-05-11:
[Uploading report_00000000007af555.zip…]()

SciresM, 2022-05-11:
@JrRamos1524 That is not a file upload, it is just self-referential blue text.

JrRamos1524, 2022-05-11:
[report_00000000007af555.zip](https://github.com/Atmosphere-NX/Atmosphere/files/8673692/report_00000000007af555.zip)

SciresM, 2022-05-11:
Issue with INI, this has been "user's console has bad RAM" in the past.

Does stock boot without using Atmosphere?

JrRamos1524, 2022-05-12:
No, I can’t turn it on, not even with the stock mood, what can I  do?

JrRamos1524, 2022-05-26:
what can I do?
help me

Masamune3210, 2022-05-26:
If the console wont boot, even without Atmosphere loaded and trying to just run stock, its a good sign that there is some kind of hardware malfunction unfortunately