Atmosphere-NX/Atmosphere #1881: Panic occured while running Atmosphere, 010041544D53000, 0xFEE

KingSlowDown, 2022-05-14:
## Bug Report

### What's the issue you encountered?

Panic occured while running Atmosphere.
Title ID: 010041544D53000
Error: std::abort <0xFFE>

### How can the issue be reproduced?

I idled in Smash Bros and it always happens after a certain amount of time passes.

### Crash Report

https://gist.github.com/KingSlowDown/30f4e8984993e8b78f8b3603d4bb3020

I also got a fatal error report, but since it's a .bin I cannot upload it to a gist nor on here. If there's a way to send it in here, tell me, that is the report the error told me it mainly produced, but the crash report got produced besides that.

### System Firmware Version

14.1.1

### Environment?

Bootloader tried with Hekate and fusee, both same outcome.
I used emuMMC and sysMMC, same outcome.
Everything up to date.

SciresM, 2022-05-14:
What you linked is useless. Upload the report binary.

yourboid, 2022-05-14:
[image]

ENjoyBlue2021, 2022-05-16:
I don't think its related to smash bros.
With latest atmosphere and 14.1.1 it just crashes after a while.
Sometimes after a couple of minutes, max. I can play 2-3h.
My emuMMC partition worked well until I updated a couple weeks ago.

hegyak, 2022-05-19:
[report_07cbcbba.zip](https://github.com/Atmosphere-NX/Atmosphere/files/8728933/report_07cbcbba.zip)
I get the Same message when attempting to Boot from Hekate 5.7.2

Using sysMMC launch option

Firmware 14.1.1

KingSlowDown, 2022-05-19:
Ah yea guys, for me it was the SD card, it was corrupted apparently. Fixed it by just using another one.

BeckysFootSlave, 2022-05-19:
If someone is giving me
Title ID: 010041544D53000
I just say that they should check if sd is really FAT32 or I tell them to check "archive bit" with hekate!