Atmosphere-NX/Atmosphere #1882: Atmosphere crashing on boot

r4ulill0, 2022-05-14:
## Bug Report

### What's the issue you encountered?

I was migrating to a bigger SD card but as I didn't update in a while I opted for a fresh install. After making the partition and creating the emuMCC in its partition, I launched the payload from hekate. The logo of atmosphere appears, then a small version logo, then the nintendo logo and finally it crashes.

### How can the issue be reproduced?

Follow [rentry guide](https://rentry.org/EristaEmuNAND) on an Erista with 5.1.0 firmware.

### Crash Report

[Report](https://gist.github.com/r4ulill0/aed3d4c79c25e561715835e82f109773)

### System Firmware Version

5.1.0</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: Hekate 5.7.2
- Official release or unofficial build: 
  - Official release version 1.3.2
- Do you have additional kips or sysmodules you're loading: No
- Homebrew software installed: N/a
- EmuMMC or SysNAND:
  - Partition based EmuMMC

### Additional context?

I have cheked the integrity of the microSD with [f3](https://fight-flash-fraud.readthedocs.io/en/latest/index.html) to check if it was a fake one but it isn't.

SciresM, 2022-05-14:
Enable nogc or eject your sd card.

You've run a firmware version which burnt newer gamecard fuses, you can never use gamecards on lower firmwares again.

closed 2022-05-14 by SciresM

r4ulill0, 2022-05-14:
I leave here some indications in case someone has the same problem:

 - Hekate autonogc option is not the one sciresM refers to and it seems to not be working as before 1.0.0 . Probably it only works if configured properly, my launch setting just load fusee.bin (maybe it works loading package3).
 - BTC.ini is no longer the way to config atmosphere, check atmosphere/config_templates/stratosphere.ini

SciresM, 2022-05-14:
Fusee doesn't use hekate configuration. The setting is described in docs/templates.