Atmosphere-NX/Atmosphere #1884: Crash boot (2105-0584)

Poop81, 2022-05-16:
## Switch error report

### What is the problem you encountered?

I bought a switch V1 a year ago on the net with the nand hs. No backup was provided to me because the person did not have to make one.

I bought a complete emmc on Aliexpress and replaced it. I followed a tutorial with a nand donor to make it work, it starts in OFW and CFW.

I wanted to update my EmuMmc from 13.0.0 to 14.1.1 and unfortunately I had a boot crash afterwards.

Stupid as I am, I reversed my sd cards, I had made a backup of my sd card with NxNandManager, but I took the wrong sd card to update.

I made various attempts by creating an Emummc from 13.0.0 and I did update by update and when I get to 14.0.0, I get a boot crash.

I make the decision to update the switch to 14.1.1 via CFW SysNand and I restart it afterwards normally so that the update is official and the efuses burn and I can boot again in CFW EmuNand the sd card which boot crash .

Unfortunately, that didn't change anything.

If I forgot to provide items tell me I will add the information.

### How can the issue be reproduced?

I start the enCFW EmuNand console, it crashes after the N* logo.
No problem if I start it in OFW

Error Code : 2105-0584 (0x49069)
Program : 0100000000000025
Firmware : 14.1.1 (Atmosphère 1.3.2)

### Incident report

https://gist.github.com/Poop81/45fe688673c54f875e8947c81f8ec5de

### System firmware version

14.1.1

### Environment?

- rocket
- Atmosphere 1.3.2
- I don't have any sysmodules or kips
- Homebrew software installed: Daybreak, reboot to payload
- Using the EmuMMC partitio
[image]
n

lsp199308, 2022-05-24:
https://github.com/CaramelDunes/prodinfo_gen/issues/24

closed 2022-05-24 by SciresM

Asenvivie, 2022-07-03:
> Hi , thank you, I had the information by this link too, and I was able to easily solve my problem :)

Hello, i have same error. Sthetix said because my nand is donor before. Try to boot via emunand and sysnand still same error result.

And his said i have to generate new prodinfo new gen. I dont understand how to made it.

Now i downgrade the fw to 13.2, it back to normal but i cant update it.

Need help guys

Poop81, 2022-07-03:
Hello, the only solution that I found is to update the normal console by going through Hekate then SysNand CFW then to start it normally so that then it is good .

Asenvivie, 2022-07-03:
> Hello, the only solution that I found is to update the normal console by going through Hekate then SysNand CFW then to start it normally so that then it is good .

Its mean that i boot to sysnand then update it from daybreak? It can be done i tried before. Error 2105-0584

Poop81, 2022-07-03:
Yes, so I also use daybreak but I made the updates one by one since 13.0.0 (the one where I was) so you take from your updates until the last one and you install them one by one

Asenvivie, 2022-07-03:
Is it work? Is your switch from donor nand before?