Atmosphere-NX/Atmosphere #1886: Daybreak causing boot hangs

424778940z, 2022-05-21:
## Bug Report

### What's the issue you encountered?

Daybreak will cause boot hangs if reset default selected during update.
Not sure if this issue only specific to emuMMC.

No there is no any change made to Atmosphere

### How can the issue be reproduced?

Format sdcard to exfat (also tested 32k 128k cluster sizes, seems no effect on the issue).

Get following release packages, then dump on the card.
https://github.com/CTCaer/hekate/releases/tag/v5.7.2
https://github.com/Atmosphere-NX/Atmosphere/releases/tag/1.3.2

Add basic boot option for emuMMC
```
[Atmo EMU]
fss0=atmosphere/package3
emummcforce=1
```
Boot into Hekate, create emuMMC, boot from emuMMC.

Use Daybreak to upgrade system, select factory reset.
Reboot, hangs at Switch logo.

### Crash Report

There is no any crash, fatal report or log produced.

### System Firmware Version

My OFW in NAND is on 7.0.1 with exFAT support, there are three users, one of them linked to Nintendo account.
Tried upgrade to ` 13.1.0, 13.2.1, 14.0.0, 14.1.0`, all stuck, seems version of target firmware have nothing todo with the issue.

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: `hekate v5.7.2 & Nyx v1.2.2`
- Official release or unofficial build: `Official Atmosphère 1.3.2`
- Do you have additional kips or sysmodules you're loading: `none`
- Homebrew software installed: `none`
- EmuMMC or SysNAND: `EmuMMC file-based`

### Additional context?

**Boot into the emuMMC, then initialize the console from system settings menu works fine, no hang after reset.**

Samsung 512G SD card, made sure no file system corruption, no error reported from chkdsk

masagrator, 2022-05-21:
> Format sdcard to exfat (also tested 32k 128k cluster sizes, seems no effect on the issue).

Don't use exfat. It's causing corruptions and using it is discouraged by Switch hacking community, Atmosphere team included.

424778940z, 2022-05-21:
> Don't use exfat. It's causing corruptions and using it is discouraged by Switch hacking community, Atmosphere team included.
> 
> Use only FAT32.
> 
> chkdsk is only checking corruption in file system, not in files itself.

I totally understand your concern, however fat32 is the same, this issue has nothing todo with exfat

SciresM, 2022-05-21:
Don't mention exFAT unless you want to be told not to use it. "Try without exFAT" will always be the best piece of advice you can be given -- it is factually that bad.

424778940z, 2022-05-21:
> Don't mention exFAT unless you want to be told not to use it. "Try without exFAT" will always be the best piece of advice you can be given -- it is factually that bad.

I'm trying to providing as much detail as possible, more details in an issue is always better, that's why I mentioned it, not to get people yelling at me.
As stated above, I also tested on fat32, no difference

SciresM, 2022-05-21:
Nobody is yelling at you. You are being told to not use exFAT because you shouldn't use exFAT.

That aside, I can't reproduce this locally.

424778940z, 2022-05-21:
> That aside, I can't reproduce this locally.

did you try it from a lower version firmware?

SciresM, 2022-05-21:
I tried while doing an upgrade from 11.X -> 13.X.

It booted successfully.

Also, for reference, we just call a system api to do the reset: https://github.com/Atmosphere-NX/Atmosphere/blob/master/troposphere/daybreak/source/ui.cpp#L1149

Not really sure what there is to be done about this even if the issue is a real one (which it may be), since we do none of the reset logic ourselves.

424778940z, 2022-05-21:
> Also, for reference, we just call a system api to do the reset: https://github.com/Atmosphere-NX/Atmosphere/blob/master/troposphere/daybreak/source/ui.cpp#L1149

when initialize the console form the setting menu, it took few mins to complete.
in daybreak only about 1 second.
are they actually the same...? or the system menu did some other things?

SciresM, 2022-05-21:
The system menu will also call some IWiper APIs on 10.0.0+. If shouldn't take a few minutes on lower firmware.

SciresM, 2022-05-21:
(And those IWiper APIs don't do anything except create and delete large files, to fill the unused space on disk with zeroes)

424778940z, 2022-05-21:
> The system menu will also call some IWiper APIs on 10.0.0+. If shouldn't take a few minutes on lower firmware.

I upgrade to 14.x then did the initialization, so that seems reasonable then.

424778940z, 2022-05-23:
I tried whole thing again, on ALL FAT32 setup, sdcard is FAT32, Daybreak chose only FAT32, and reset, same boot issue, hang on logo forever.
Either I do the reset from 7.0.1 then upgrade, or I upgrade to 14.0.0 then do reset, both way works fine. **<--I mean the system menu initialization.**

SciresM, 2022-05-23:
I mean, I don't know what to tell you. It may be an issue when going specifically from version X -> version Y. It pretty much works fine here, though I haven't tried 7.0.1 -> 14.x.

We use the reset to factory settings for refurbishment system API.

Settings menu calls reset or reset without user savedata; refurbishment wipes more stuff.

Honestly, there's kind of nothing that we can really do, since we're just calling a system API. If you have a workaround for that specific version upgrade on your device, I would use it.

closed 2022-05-23 by SciresM

putnam, 2022-08-13:
I reproduced this same issue tonight. System was on 14.1.2. Created EmuMMC. Booted EmuMMC and reinstalled 14.1.2 with Daybreak, telling it to initialize to factory settings. EmuMMC now hangs on every boot with the "Nintendo Switch" logo. Also attempted getting into maintenance mode with vol+/-, and with system settings config in atmosphere. No dice with either.

Always FAT32 for me.

424778940z, 2023-07-19:
> The reason I needed to do this was related to Nintendo Online accounts that were blocking a reset the normal way, due to lack of connectivity.

Just walking through my old emails, maybe late to the party, and may off the topic.
But it's worth to mention, some homebrews could remove it, like [Goldleaf](https://github.com/XorTroll/Goldleaf).