Atmosphere-NX/Atmosphere #1887: Fatal error std: abort when trying to run FSS0SYS

EdgarUstian, 2022-05-24:
## Bug Report

Atmosphere produces a fatal error on boot and refuses to start. 

### What's the issue you encountered?

I updated to the latest version of Atmosphere and when trying to run it like how I usually do it, it produces fatal error on boot from Hekate. No changes were made, simply updated to the latest release.

### How can the issue be reproduced?

? Update to latest atmo and hekate/nyx and then run the CFW

[ * ]</br>
[ Include a detailed step-by-step process for recreating your issue. ]</br>

### Crash Report

Fatal Report
[01652048707_0100000000001000.log](https://github.com/Atmosphere-NX/Atmosphere/files/8759890/01652048707_0100000000001000.log)

### System Firmware Version

14.1.1</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: Hekate
- Official release or unofficial build: 5.7.2
- Do you have additional kips or sys modules you're loading: N/A
- Homebrew software installed: N/A
- EmuMMC or SysNAND: N/A

masagrator, 2022-05-24:
> * Do you have additional kips or sys modules you're loading: N/A
> * Homebrew software installed: N/A

well, those are pretty important as error suggests that there is a conflict with home menu - aka qlaunch.

SciresM, 2022-05-24:
Tentatively: remove custom theme, etc. Qlaunch errors are almost *always* this.

Also, yeah, you put N/A for fields that are deeply applicable.

Also, Hekate isn't made by us.

Anyway, closing on extremely high probability that this is a user-custom-theme-or-config problem. 

Please re-open this issue with more environment details if the issue continues after deleting the atmosphere/contents folder.

SciresM, 2022-05-24:
Also, wait, that's a fatal report, not a fatal error binary.

If the error screen is reporting a fatal error/std::abort (and the screen is black/in hekate), you uploaded the wrong file.

If you re-open, please post a picture of the screen displayed and make sure you upload the correct file mentioned in the display.

closed 2022-05-24 by SciresM

EdgarUstian, 2022-05-24:
alright let me make sure im looking at the correct file

EdgarUstian, 2022-05-25:
Alright I found the actual file I thought the most recent one will be the right one but it wasnt 

[image]

I am having trouble uploading the bin file itself though

SciresM, 2022-05-25:
I will say, given "Failed to parse" I'm willing to bet you just have SD card corruption. Not to ask an obvious question or anything, but have you tried deleting the atmosphere directory/replacing with unmodified clean zip? (And have you tried reformatting your sd card?)

EdgarUstian, 2022-05-25:
Failed to parse I fixed because I ran it off the wrong payload.
I want to try deleting rather than replacing but i dont know if any of game specific data will get lost?

I checked and there was no game specific folder there, so I took your advice and deleted atmo folder and pasted in a new one

that fixed the issue. Thank you