Atmosphere-NX/Atmosphere #1888: Won't Open Atmosphère After Injecting Payload

Omega207, 2022-05-24:
## Bug Report
**NOTE!** This may not be a "bug report", but rather a issue with my system in general. If it's related to TegraRCM itself, please redirect me to that issue page.<br />

After injecting my payload with TegraRCMGUI (which it says it injected successfully), I have to boot into Atmosphère to proceed to the next step. However, no matter what I do, the menu(s) will not load. I would pass this off as my system being unhackable, but since the payload injection was successful, I think it might be something else.

### What's the issue you encountered?

As mentioned above, Atmosphère will not boot via any known methods. The Switch will simply boot normally.

### How can the issue be reproduced?

_Because this may be an issue with my own Switch, no reproduce methods will be shown._

### Crash Report

[N/A]

### System Firmware Version

14.1.1

### Environment?

- Exploit: Fusèe
- Release type: Official release (See FWV above)
- Additional kips or sysmodules: No
- Homebrew software installed: None except nessesary
- EmuMMC or SysNAND: None

### Additional context?

If you need screenshots for payload confirmation, please confirm I can attempt to re-inject the payload. <br />
My console is "patched", according to ismyswitchpatched, but it couldn't find the firmware version.

Anything relating to this question helps (**Please, I seriously don't want to buy another Switch!**),
Omega207

SciresM, 2022-05-24:
Honestly this seems like a support question. Join the Nintendo homebrew or ReSwitched discords for support -- this issue tracker is for bugs.

closed 2022-05-24 by SciresM

Omega207, 2022-05-24:
@SciresM  Sorry, for unexplainable reasons, I cannot join any sort of Discord.

SciresM, 2022-05-24:
I'm sorry, but you'll have to find a venue that isn't this issue tracker. If you can't join a discord, post in a reddit help thread or something.