Atmosphere-NX/Atmosphere #1889: NX-Shell crash, possibly related to number of subfolders

ToxiClay, 2022-05-27:
## Bug Report

### What's the issue you encountered?

I encountered a repeatable crash in AMS 1.3.2 on HOS 14.1.1.

### How can the issue be reproduced?

Prerequisite: [NX-Shell](https://github.com/joel16/NX-Shell/releases).

Steps:

1. Open NX-Shell in Applet Mode.
2. Navigate to \Nintendo\Contents.
3. Click on "registered."
4. Atmosphere crashes.

Note: This *does not occur* when launching the homebrew menu in high-memory, Title Override mode. Might this have to do with some sort of memory overflow when reading a very large folder? There are 248 folders inside my "registered" folder.

### Crash Report

[Atmosphere NX-Shell 0x4a8](https://gist.github.com/ToxiClay/58e41a9d938469fcb9eb42cb014e6046)

[Atmosphere NX-Shell 0x0](https://gist.github.com/ToxiClay/fe9dc5d5d09792ace540feb09a648d57)

### System Firmware Version

14.1.1

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: Fusee, chainloaded through Hekate.
- Official release or unofficial build: Official AMS 1.3.2
- Do you have additional kips or sysmodules you're loading: Sys-ftpd-light and sys-dvr
- Homebrew software installed: 
```
* 90DNS Tester
* AIO Switch Updater
* Appstore
* Breeze
* Browser
* DBI
* Edizon
* Goldleaf
* NXDumpTool
* NX-Shell
* Payload Launcher
* Untitled
* Daybreak
* Reboot To Payload
* Sys-DVR Configuration Utility
```

- EmuMMC or SysNAND:
  SysNAND

### Additional context?

N/A

SciresM, 2022-05-27:
This isn't an atmosphere crash -- this is a problem with the homebrew you're using.

You should report it to the developer of NX-Shell.

closed 2022-05-27 by SciresM

ToxiClay, 2022-05-27:
My apologies. :)