Atmosphere-NX/Atmosphere #1891: Super Mario Striker: Battle League Football First Kick crash on start

Fuffelpups, 2022-05-28:
### What's the issue you encountered?
Super Mario Striker: Battle League Football First Kick crashes every time.

I can play the game, when i disable Atmosphère.

### Crash Report
https://gist.github.com/Fuffelpups/a2d42297f0a2dfd25df272dac628baba

### System Firmware Version
14.1.1|AMS 1.3.2|S</br>

### Environment
- bootloader hekate
- SysNAND

SciresM, 2022-05-28:
What custom sysmodules, if any?

post a screenshot of your atmosphere/contents directory?

Fuffelpups, 2022-05-28:
I disabled all my custom s

> What custom sysmodules, if any?
> 
> post a screenshot of your atmosphere/contents directory?

Now i feel stupid. I made a quick fresh install, because there were a lot of folders and it works now. Taking screenshots would have taking way longer. Thanks and sorry. I should have tried this first.

SciresM, 2022-05-28:
Yeah, this is usually how it goes. No worries.

closed 2022-05-28 by SciresM