Atmosphere-NX/Atmosphere #1892: Error on Boot 0x4A2 (2162-0002)

JerneV, 2022-05-28:
## Bug Report

### What's the issue you encountered?

From Hekate boot into EmuMMC with Atmosphère 1.3.2. But received a crash and switch goes back to Hekate. 
Updated Atmosphère to 1.3.2, as well as fusee.bin as well as hekate itself. 

### How can the issue be reproduced?

From Hekate boot into Atmosphère. 

### Crash Report

Atmosphère Fatal Report (v1.1):
Result:                          0x4A2 (2162-0002)

Program ID:                      010000000000100c
Process Name:                    overlayDisp
Firmware:                        13.2.0 (Atmosphère 1.3.2-master-e96972c93)
General Purpose Registers:
         FP:                     00000068fd42b630
         LR:                     0000002d08bab2e4
         SP:                     00000068fd42b630
         PC:                     0000002d08b4b284
Start Address:                   0000002d08a00000
Stack Trace:
        ReturnAddress[00]:       0000002d08bf3f7c
        ReturnAddress[01]:       0000002d08bf3d48
        ReturnAddress[02]:       0000002d08bf3924
        ReturnAddress[03]:       0000002d08bf3a78
        ReturnAddress[04]:       0000002d08b1686c
        ReturnAddress[05]:       0000002d08b15d30
        ReturnAddress[06]:       0000002d08b15e98
        ReturnAddress[07]:       0000002d08a01ed8
        ReturnAddress[08]:       0000002d08d59188
        ReturnAddress[09]:       0000002d08d83cec
        ReturnAddress[10]:       0000002d08c01524
        ReturnAddress[11]:       0000002d08c1b344
        ReturnAddress[12]:       0000002d08c1ab8c
        ReturnAddress[13]:       0000002d08b0c0f8
        ReturnAddress[14]:       0000002d08b0f2a4
        ReturnAddress[15]:       0000000000000000
Stack Dump:                               00 01 02 03 04 05 06 07 08 09 0a 0b 0c 0d 0e 0f
                             0068fd42b630 d0 b6 42 fd 68 00 00 00 7c 3f bf 08 2d 00 00 00
                             0068fd42b640 01 01 00 00 00 00 10 00 0c 00 00 80 01 00 00 00
                             0068fd42b650 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
                             0068fd42b660 05 00 00 00 00 00 00 00 00 00 00 00 08 00 00 00
                             0068fd42b670 00 00 00 00 14 00 00 00 b0 87 bb 08 01 00 00 00
                             0068fd42b680 88 b6 42 fd 68 00 00 00 00 00 00 00 00 00 00 00
                             0068fd42b690 00 00 a0 08 2d 00 00 00 00 00 00 00 00 00 00 00
                             0068fd42b6a0 00 00 00 00 00 00 00 00 15 00 00 00 00 00 00 00
                             0068fd42b6b0 e0 93 fc 08 2d 00 00 00 50 02 00 00 00 00 00 00
                             0068fd42b6c0 40 b6 42 fd 68 00 00 00 00 16 d7 ed 76 00 00 00
                             0068fd42b6d0 20 b7 42 fd 68 00 00 00 48 3d bf 08 2d 00 00 00

### System Firmware Version

OFW is 14.0
CFW is 13.2</br>

### Environment?

- Hekate
- Official release 
- EmuMMC (partition)

SciresM, 2022-05-28:
This one is almost certainly a user setup issue. 

Post a screenshot of your atmosphere/contents dir?

JerneV, 2022-05-28:
[image]

It's a fresh install of Atmos, so there isn't a contents folder

SciresM, 2022-05-28:
Hm. Okay, that's unexpected.

* Does the issue occur on sysmmc?
* Does the issue occur booting using fusee?
* Could you upload the full report file?

I can't really think of what this could be on a stock configuration, to be honest...

SciresM, 2022-05-28:
Are you sure you're uploading the correct report file, and that this isn't a fatal error (black screen with text)?

JerneV, 2022-05-28:
- Booting Stock is no issue at all
- Just tried booting with fusee, results in the exact same issue as far as I can see
- And yes 
[Full Report Files.zip](https://github.com/Atmosphere-NX/Atmosphere/files/8791198/Full.Report.Files.zip)

JerneV, 2022-05-28:
> Are you sure you're uploading the correct report file, and that this isn't a fatal error (black screen with text)?

Nope, this is the type of screen I get (picture from the internet, not actually my error screen) 
[image]

SciresM, 2022-05-28:
Er, I wasn't asking about stock. I meant atmosphere sysmmc.

JerneV, 2022-05-28:
Seems to work just fine too. 

[image]

SciresM, 2022-05-28:
If sysmmc works, my *inclination* is to say that there's some form of SD card corruption or other SD card issues.

Kind of a pain in the ass, but you might want to try e.g. reformatting your SD, trying another SD card, etc?

JerneV, 2022-05-28:
Haha, that _is_ a real pain in the ass. But yeah, a reinstall will probably do the trick.