Atmosphere-NX/Atmosphere #1893: Atmophere makes Ultimate crash

R-son, 2022-05-28:
According to my error logs, Atmosphere crashed. I have no idea why it started crashing, I didn't do anything and it suddenly crashed. I sent the log file. I'm also sending the error on the switch screen.
[01653754315_0100000000000023.log](https://github.com/Atmosphere-NX/Atmosphere/files/8791467/01653754315_0100000000000023.log)
[image]

R-son, 2022-05-28:
[error_log.txt](https://github.com/Atmosphere-NX/Atmosphere/files/8791472/error_log.txt)

SciresM, 2022-05-28:
Reopen using the issue templates.

closed 2022-05-28 by SciresM

R-son, 2022-05-28:
Where is it ?

Masamune3210, 2022-05-29:
Might want to go ahead and just make a new issue with the template filled out instead of editing this one retroactively