Atmosphere-NX/Atmosphere #1898: Mario Strikers crashes on file based emummc

josete2k, 2022-06-08:
###` What's the issue you encountered?

Game crashes after te loading Screen, I have to force power off the console</br>
only on emummc file based (resized), same config boots ok on sysmmc (game installed in SD, not in the sysmmc) and ok in hidden partiton emummc.</br>
Tying to disable sys-clk, emuiibo, ediZon and mission control and same result, sysmmc is working even they're actived</br>
If I active Plane mode, joycons are unrecognized just before the crashing</br>

### Crash Report

### System Firmware Version

14.1.1</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: hekate (fss0) and fusee.bin, same result.
- Official release or unofficial build: 1.3.2
- Do you have additional kips or sysmodules you're loading: Sys-clk, emuiibo, ediZon and mission control
- Homebrew software installed: [ * ]
- EmuMMC or SysNAND:
  - Resized emummc file based

### Additional context?

- Additional info about your environment: Sysnand is clean (no games installed) and emummc has more than 50 installed titles.
- Chache cleared by accessing to te HOS recovery.
- Orphan files removed by DBI

SciresM, 2022-06-08:
Sorry, piracy is not supported. Please do not open duplicate issues when your issue is closed for piracy; you'll have to seek support elsewhere.

closed 2022-06-08 by SciresM

josete2k, 2022-06-08:
Sorry and thanks again.