Atmosphere-NX/Atmosphere #1899: Console softlocks whe tring to open applet

Alexlin465, 2022-06-11:
## Bug Report

Atmosphere crashed/soflocked while trying to open applet from album

### What's the issue you encountered?

I was trying to enter applet after opening Pokémon Shining Pearl and when I did, the screen was stuck after pressing A. I was able to power off, but I think it would be a problem after a long time.

### How can the issue be reproduced?

Opening a game, going to HOME menu then entering applet from album.

### Crash Report

None, it just leaves the screen hanging

### System Firmware Version

14.1.1

### Environment?

- Bootloader: Hekate
- Atmosphere ver 1.3.2
- Homebrew software installed: Super Mario 64 v8

SciresM, 2022-06-11:
Post a screenshot of your atmosphere/contents dir?

StickIT, 2022-06-13:
I have the same problem.
[01655130711_010000000000100d.log](https://github.com/Atmosphere-NX/Atmosphere/files/8892002/01655130711_010000000000100d.log)

StickIT, 2022-06-13:
Nothing on atmosphere/contents

closed 2022-07-06 by SciresM