Atmosphere-NX/Atmosphere #1900: Black screen on boot

scrappysphinx, 2022-06-12:
Hi all, i have a Switch Lite fitted with an SX Lite board. It was originally bought with an SX OS License but i have reflashed it with spacecraft nx. It will boot hekate and then i can boot into Atmosphere using fusee from there but i cannot boot fusee (named payload.bin on the root of the sd card) directly i just get a black screen like the switch isn't even powering up. Any ideas?

Thanks

H4X0R46, 2022-06-12:
They're gonna tell you to use the template and close this thread 👀

As for your actual issue, I'm a bit confused. Booting Hekate would mean that the Hekate payload is on your SD card as payload.bin. booting straight into atmosphere would mean that ATMOSPHERE's payload would he be payload.bin?

So I'm thinking, the best way would be to use the setting inside hekate to autoboot the atmosphere payload, that's the same outcome and much simpler 🤔

SciresM, 2022-06-12:
You'd be correct, that I'm going to say "use the issue templates.

But I'm also going to say that this sounds like a support question, not a bug, and that the issue tracker is for bugs, not for support.

Seek support on discord.

closed 2022-06-12 by SciresM