Atmosphere-NX/Atmosphere #1901: Problems using 2 separate loops in 2 cheat code

ImanCol, 2022-06-13:
## Problem using Loop in cheat code

I am using a 4 loop loop.

I have a Loop in each 1 of the 4 cheat codes I made.  The thing is that only 1 works correctly, it does 4 loops, but the others only have 3 loops despite having 4.

SciresM, 2022-06-13:
This sounds like a support issue and not like a bug. Also, you didn't use the issue templates.

Come seek support on discord?

SciresM, 2022-06-13:
There's also nowhere near enough detail here.

closed 2022-06-13 by SciresM

ImanCol, 2022-06-13:
It doesn't matter, apparently the loops add from the first one, I thought the first one kept R0 at 0. So it's not 0,1,2,3 but 1,2,3,4

I don't know why you usually close problems like that without looking for a solution.  In this case, it is necessary to document more how it works.

ImanCol, 2022-06-13:
but I keep experiencing the error that one works and others do not 🤷🏻‍♂️

SciresM, 2022-06-13:
> I don't know why you usually close problems like that without looking for a solution.

Because the issue tracker is for bugs, not for support, and this is a support problem.

I have told you that this isn't a bug, and to seek support on discord. The issue tracker is for bugs, and I don't believe it likely this is a bug.

You also ignored the template for issues.

SciresM, 2022-06-13:
If you would like to get further help, I am happy to talk more to you about the issue *on discord*.