Atmosphere-NX/Atmosphere #1902: Invalid GPT Signature

j3rkf4ce, 2022-06-14:
## Bug Report

### What's the issue you encountered?

"A Fatal error occured when running Fusee"
"Invalid GPT signature" 

"Press POWER to reboot."

### How can the issue be reproduced?

Running Latest Fusee/Atmosphere Version
Running Fusee through Tegra RCM GUI
Error Screen Appearing

### Crash Report

N/A
Crash Report Never Generated

### System Firmware Version

13.1.0

### Environment?

- Fusee payload (Latest Version)
- Official Release 1.3.2
- N/A
- Tinfoil
- emuMMC (Partition)

### Additional context?

- Occured when updating atmosphere

SciresM, 2022-06-14:
This sounds like you have a corrupt MMC. Does the console boot ofw without using hacks?

SciresM, 2022-06-14:
That is extremely surprising to me. Does the console boot using hekate instead of fusee?

SciresM, 2022-06-14:
Oh, wait, emummc. I assumed this was sysmmc. Your emummc is probably corrupt. Have you tried making a fresh one?

j3rkf4ce, 2022-06-14:
> Oh, wait, emummc. I assumed this was sysmmc. Your emummc is probably corrupt. Have you tried making a fresh one?

I'll try to backup my files on my emuMMC and make a clean slate

j3rkf4ce, 2022-06-15:
Ok, I got a clean emuMMC and the error appears to be gone. Only issue now is just getting my games/save files back I guess.

closed 2022-06-15 by j3rkf4ce