Atmosphere-NX/Atmosphere #1903: panic occurred while running atmosphere.

Burritoman101, 2022-06-16:
## Bug Report
N/A
[ If any section does not apply, replace its contents with "N/A". ]</br>
[ Lines between [ ] (square brackets) should be removed before posting. ]</br>

[ * ]</br>
[ Note: If the bug or crash you encountered is related to; ]</br>
[ - software used to make "backups", ]</br>
[ - software explicitly distributed for piracy, etc ]</br>
[ then contributors will not provide support for your issue and your issue will be closed. ]</br>

### What's the issue you encountered?
The following error appears when trying to boot from Hekate atmosphere
Panic occurred while running Atmosphere
Title ID: 010041544d530000
Error:       std::abort (0xFFE)

I didnt make any chnages to Atmosphère
[ Describe the issue in detail and what you were doing beforehand. ]</br>
[ Did you make any changes related to Atmosphère itself? ]</br>
[ If so, make sure to include details relating to what exactly you changed. ]</br>

### How can the issue be reproduced?
Boot to RCM and launch Hekate and then Boot Atmosphere emunand
[ * ]</br>
[ Include a detailed step by step process for recreating your issue. ]</br>

### Crash Report
Not a crash report but a Fatal Error
[ Crash reports can be found under ``/atmosphere/crash_reports``. ]</br>
[ If your issue caused Atmosphère to crash, include the crash report(s) by creating a [gist](https://gist.github.com/) and pasting the link here. ]</br>https://gist.github.com/Burritoman101/a354d7b5a94bea2c00c94f9efb2ae98c
[ If you don't include a crash report in instances of crash related issues, we will ask you one to provide one. ]</br>

### System Firmware Version

14.1.1</br>
[ Replace X's with system firmware version at time of crash. ]</br>
[ You can find your firmware version in the Settings -> System, under "System Update". ]</br>
[ If it says "Update Pending", you can clear the pending update by rebooting to Maintenance Mode. ]</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: hekate
- Official release or unofficial build:  unofficial
  - [ Official release version x.x.x (or) unofficial build ]
  - [ If using an unofficial build, include details on where/how you acquired the build. ] 
     - [ Ex: Self-compilation ]
     - [ Ex: Kosmos' distribution of Atmosphère ]
- Do you have additional kips or sysmodules you're loading:N/A
- Homebrew software installed: [ N/A]
- EmuMMC or SysNAND: EmuMMC file-based
  - [ If using an EmuMMC, include whether it's partition-based or file-based. ]

### Additional context?

- Additional info about your environment:
- [ Any other information relevant to your issue. ]

Masamune3210, 2022-06-16:
Gathered from other issues with same titleid, I'm going to hazard a guess and say something is up with your PRODINFO

Masamune3210, 2022-06-16:
Maybe try making sure your sd card isn't corrupted and maybe test recreating your emunand?

SciresM, 2022-06-23:
The report you uploaded appears to just be a filename and not an actual report.

Also, your issue says it's an unofficial build. Closing this, re-open if it occurs with an official build, and actually upload the report.

closed 2022-06-23 by SciresM