Atmosphere-NX/Atmosphere #1908: Issues with 64GB cards, but not with 128GB and 16GB cards

HowlingWolfHWC, 2022-06-23:
## Bug Report

### What's the issue you encountered?

Atmosphère 1.3.2 crashes in a NSW with OFW 13.2.1, SysNAND when using two different 64GB microSD cards (one ADATA, one SanDisk, one exFAT, the other one was also exFAT, but I tried it in FAT32 too afterwards, formatted with rufus). I've tried also a 16GB microSD (FAT32) card and a 128GB microSD card (exFAT) and they worked with no issues.
So, the crash occurs ONLY when using 64GB microSD cards.

### How can the issue be reproduced?

Run the latest Atmosphère version in a console with 13.2.1 using a 64GB microSD card.

### Crash Report

I have no crash_reports, but I do have some fatal_errors:
[atmosphere_errors_64gbmicrosd.zip](https://github.com/Atmosphere-NX/Atmosphere/files/8964715/atmosphere_errors_64gbmicrosd.zip)

### System Firmware Version

13.2.1

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: fusèe
- Official release or unofficial build: Official release 1.3.2
- Do you have additional kips or sysmodules you're loading: not at all, it is a vanilla Atmosphère with sigpatches only (which works flawless using cads with other sizes)
- Homebrew software installed: DBi, JKSV
- EmuMMC or SysNAND: SysNAND

### Additional context?

- Additional info about your environment:
Working cards: 128GB SanDisk (exFAT), 16GB Kingston (FAT32)
Crashing cards: 64GB SanDisk (exFAT, FAT32), 64GB ADATA (exFAT)
The entire contents of the microSD was copypasted from the 64GB SanDisk to every other cards, so there's no corruption in the files

HowlingWolfHWC, 2022-06-23:
On a side note:
The two 64GB microSD cards do work flawlessly on other devices, both are legit and properly tested. So, yeah, it's not the card's fault.

SciresM, 2022-06-23:
1) fs::ResultPortSdCardNoDevice() is a hardware result code in Nintendo's FS driver code which cannot result from Atmosphere. In particular, it occurs when Nintendo's sdmmc::DeviceDetector::IsInserted() returns false.
2) It may or may not be your card, but keep in mind it may be your SD card reader. All I can tell you for sure is that this is a hardware problem, not an atmosphere software problem.
3) Piracy is not supported,

closed 2022-06-23 by SciresM

HowlingWolfHWC, 2022-06-23:
1 - New fun fact out of my new tests: Both 64GB microSD cards works with Atmosphère 1.2.6, so, Atmosphère is the one to blame here.
2 - Not the case as seen above
3 - I'm not into pirating, but I like my forwarders to the softwares I use in the main menu next to my games.

SciresM, 2022-06-23:
1. Sorry, but again -- there is no mechanism by which atmosphere code can cause fs::ResultPortSdCardNoDevice().
2. Is the case, per above.
3. Piracy is not supported.