Atmosphere-NX/Atmosphere #1914: A fatal error while running atmosphere ID:010041544d530000

BASgamer, 2022-07-03:
## Bug Report

### What's the issue you encountered?

The switch was booting up fine until i updated atmosphere to the latest release then after that when i push the payload i get a fatal error. 

A fatal error occurred when running Atmosphére.
Title ID: 010041544d530000
Error Desc: std::abort() called (0xffe)

Was on a 13.x firmware and when updating the atmosphere files. Got a crash then.
Used older atmosphere again and it booted correctly.
Thought maybe updating the firmware would help so updated switch to 14.1.2 and updated atmosphere to the latest and got the crash again.

### How can the issue be reproduced?

Pushing Fusee.bin with tegraRCM

### Crash Report

[report_0000000005c22b52.zip](https://github.com/Atmosphere-NX/Atmosphere/files/9035292/report_0000000005c22b52.zip)

### System Firmware Version

14.1.2

### Environment?

Pushing Fusee.bin with tegraRCM
Atmosphère 1.3.2
No kips or sysmodules
Nothing in the contents folder
SysNAND

### Additional context?

Ive tried multiple things to fix it
Removing the content folder
reformating the sd
tried with a different sd and still crashes
archive bit in hekate
had incognito rcm on and restoring that

BASgamer, 2022-07-04:
on a whim i tried 1.3.1 atmosphere and that worked... so if any help figuring out why 1.3.2 didnt would still be helpful so it wont be an issue in the future would be nice

BASgamer, 2022-07-04:
https://gbatemp.net/threads/error-010041544d530000-when-booting-atmosphere-cfw.590522/ 
solved my issue guess for some switches power_menu_reboot_function = str!payload crashes it?

SciresM, 2022-07-06:
`settings::InvalidFormatSettingsItemValue` just indicates malformed user-configuration INI. I can look at the file itself if you want to upload it, but not a bug.

closed 2022-07-06 by SciresM

BASgamer, 2022-07-08:
[system_settings.zip](https://github.com/Atmosphere-NX/Atmosphere/files/9074320/system_settings.zip)
think this was it

Masamune3210, 2022-07-08:
Maybe your editor is inserting some invisible characters and it's confusing the parser? Try recreating the issue using a different text editor

ObscenityIB, 2024-11-22:
necro, just had this same issue, I was using VSCodium on Linux to edit the text files.