Atmosphere-NX/Atmosphere #1917: Randomly got this Crash while loading a theme in Theemzer

lando07, 2022-07-10:
Here's the log, I thought I'd let yall look thru it because I don't know what went wrong but I'm sure ya'll can use it to fix bugs for the next release
[01657382859_010000000000100d.log](https://github.com/Atmosphere-NX/Atmosphere/files/9079826/01657382859_010000000000100d.log)
e

Masamune3210, 2022-07-10:
Go back and fill out the issue template please, also this log just says that the homebrew died which caused hbloader to crash

lando07, 2022-07-10:
Where do I find it?

Masamune3210, 2022-07-10:
It would have been pre-populated in the issue when you first created it

lando07, 2022-07-10:
Strange, It didn't do that for me, I'm going to repost it

masagrator, 2022-07-10:
Looks like memory leak - either because of homebrew bug or using it in applet mode when it requires more ram than applet mode gives.
Not Atmosphere related

lando07, 2022-07-10:
Oh yea I was in applet mode

closed 2022-07-14 by SciresM