Atmosphere-NX/Atmosphere #1918: A crash while playing animal crossing: new horizons

nbdeity, 2022-07-14:
only after i installed sys-botbase did this start happening.
I am not entirely sure what any of this means, but any help would be greatly appreciated.

Atmosphère Fatal Report (v1.1):
Result:                          0x265 (2101-0001)

Program ID:                      0100000000000010
Process Name:                    ptm
Firmware:                        14.1.2 (Atmosphère 1.3.2-master-e96972c93)
General Purpose Registers:
         FP:                     00000027d042abb0
         LR:                     0000005f7f8229c4
         SP:                     00000027d042abb0
         PC:                     0000005f7f82447c
Start Address:                   0000005f7f800000
Stack Trace:
        ReturnAddress[00]:       0000005f7f83d754
        ReturnAddress[01]:       0000005f7f83d428
        ReturnAddress[02]:       0000005f7f83d004
        ReturnAddress[03]:       0000005f7f83d158
        ReturnAddress[04]:       0000005f7f80b9cc
        ReturnAddress[05]:       0000005f7f80b750
        ReturnAddress[06]:       0000005f7f80b81c
        ReturnAddress[07]:       0000005f7f84fbbc
        ReturnAddress[08]:       0000005f7f84f9d8
        ReturnAddress[09]:       0000005f7f851ad8
        ReturnAddress[10]:       0000005f7f851538
        ReturnAddress[11]:       0000005f7f815298
        ReturnAddress[12]:       0000005f7f817044
        ReturnAddress[13]:       0000000000000000
Stack Dump:                               00 01 02 03 04 05 06 07 08 09 0a 0b 0c 0d 0e 0f
                             0027d042abb0 50 ac 42 d0 27 00 00 00 54 d7 83 7f 5f 00 00 00
                             0027d042abc0 01 01 00 00 00 00 10 00 0c 00 00 80 01 00 00 00
                             0027d042abd0 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
                             0027d042abe0 05 00 00 00 00 00 00 00 00 00 00 00 08 00 00 00
                             0027d042abf0 00 00 00 00 14 00 00 00 80 3d 82 7f 01 00 00 00
                             0027d042ac00 08 ac 42 d0 27 00 00 00 00 00 00 00 00 00 00 00
                             0027d042ac10 00 a0 42 d0 27 00 00 00 00 00 00 00 00 00 00 00
                             0027d042ac20 00 00 00 00 00 00 00 00 15 00 00 00 00 00 00 00
                             0027d042ac30 b0 3f 91 7f 5f 00 00 00 50 02 00 00 00 00 00 00
                             0027d042ac40 c0 ab 42 d0 27 00 00 00 00 1a 8c ff 2b 00 00 00
                             0027d042ac50 a0 ac 42 d0 27 00 00 00 28 d4 83 7f 5f 00 00 00
                             0027d042ac60 1c b8 80 7f 5f 00 00 00 c0 ae 42 d0 27 00 00 00
                             0027d042ac70 34 8d 89 7f 5f 00 00 00 00 00 00 00 00 00 00 00
                             0027d042ac80 d0 a5 8c 7f 5f 00 00 00 08 ad 42 d0 27 00 00 00
                             0027d042ac90 00 00 00 00 00 00 00 00 c0 a4 8c 7f 5f 00 00 00
                             0027d042aca0 d0 ac 42 d0 27 00 00 00 04 d0 83 7f 5f 00 00 00
TLS Address:                 0000002bff8c1a00
TLS Dump:                                 00 01 02 03 04 05 06 07 08 09 0a 0b 0c 0d 0e 0f
                             002bff8c1a00 06 00 10 00 0c 00 00 80 01 00 00 00 83 80 a3 33
                             002bff8c1a10 00 00 00 00 50 02 00 00 b0 3f 91 7f 14 00 00 f0
                             002bff8c1a20 53 46 43 49 01 00 00 00 02 00 00 00 00 00 00 00
                             002bff8c1a30 65 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00
                             002bff8c1a40 58 ad 42 d0 27 00 02 00 00 01 00 00 00 00 00 00
                             002bff8c1a50 c0 ac 42 d0 27 00 00 00 40 01 09 c1 01 00 00 00
                             002bff8c1a60 00 00 00 00 00 00 00 00 00 00 00 00 02 00 00 00
                             002bff8c1a70 00 00 00 00 0a 00 00 00 9c ad 42 d0 27 00 00 00
                             002bff8c1a80 02 00 00 00 00 00 00 00 78 ac 42 d0 27 00 00 00
                             002bff8c1a90 00 1a 8c ff 2b 00 00 00 50 ad 42 d0 27 00 00 00
                             002bff8c1aa0 78 3d 84 7f 5f 00 00 00 00 00 00 00 00 00 00 00
                             002bff8c1ab0 21 00 00 00 2b 00 00 00 00 00 00 00 00 00 00 00
                             002bff8c1ac0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
                             002bff8c1ad0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
                             002bff8c1ae0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
                             002bff8c1af0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00

SciresM, 2022-07-14:
1) You didn't use the issue templates
2) If the issue doesn't occur without extra sysmodules, I don't think it's an atmosphere bug -- and this tracker is for ams bugs, not for support.

closed 2022-07-14 by SciresM

nbdeity, 2022-07-14:
ok, i'm really sorry. i forgot to look for everything