Atmosphere-NX/Atmosphere #1921: Third Party Bluetooth Controller Not Connecting

Need-Help-T-T, 2022-07-18:
## Bug Report

### What's the issue you encountered?
I have an Xrocker Elite controller that connects to my OFW Switch but not to any of my two CFW Switches. 

On the OFW it connects both wired and wirelessly.
On the CFWs neither method work.

The CFW Switches work with all of the Joy-Cons and have no pairing issues there. I have also double checked my settings regarding allowing pro wired controller and bluetooth connection as well, both are on. 

After reading #1704 and #1705 I removed all sysmodules (tesla overlay) and it still does not connect.

I use hekate sigpatches. I'm not sure if that is classed as a modification to Atmosphere. I just copy+paste and nothing else.

All three consoles are on the latest firmware. I have latest atmosphere and use TegraGUI 2.6 to inject fusee.bin and then usually keep it in CFW until I need to access the MicroSD for FW updates.
I also have latest hekate installed and inject 5.8 if I need to access those settings. 
If in Hekate, I reboot to RCM and then inject fusee.bin

### How can the issue be reproduced?

Attempt to connect third party controller to switch.

### Crash Report

N/A

### System Firmware Version

14.1.2
AMS 1.3.2

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:
TegraGUI 2.6 to inject fusee.bin (from latest official release)
Hekate official build 5.8 when Hekate is needed.
If in Hekate, I reboot to RCM and then inject fusee.bin

- Official release or unofficial build: 
 Both are official builds

- Do you have additional kips or sysmodules you're loading:
sigpatches from [gbatemp](https://gbatemp.net/threads/sigpatches-for-atmosphere-hekate-fss0-fusee-package3.571543/)

- Homebrew software installed:
Tinleaf
Daybreak
Goldleaf
JKSV
nxdumptool
Reboot to Payload
Tinleaf Installer

- EmuMMC or SysNAND:
CFW is in EmuMMC (file based)
  

### Additional context?
Each update, I copy and paste the updated files. I haven't removed any old settings/files except for the update that required fusee_primary to be removed and replaced with fusee. I guess I was worried of losing save files or corrupting the CFW or softlocking myself.

masagrator, 2022-07-18:
> I use hekate sigpatches.

Piracy is not supported

Need-Help-T-T, 2022-07-18:
> > I use hekate sigpatches.
> 
> Piracy is not supported

I understand. But I'm not asking how to pirate. I am asking how to make the controller connect...

I mentioned since it asks for patches or modifications to atmosphere

SciresM, 2022-07-18:
I understand you're not asking how to pirate.

unfortunately, piracy is not supported -- support is not provided to users engaging in piracy, as a matter of policy.