Atmosphere-NX/Atmosphere #1926: Fatal Error Report

Mexyican, 2022-07-28:
Heres the error i dont know to how to fix this any ideas? 
here the report.

Atmosphère Fatal Report (v1.1):
Result:                          0xF601 (2001-0123)

Program ID:                      0100000000000081
Process Name:                    btred
Firmware:                        14.1.2 (Atmosphère 1.3.2-master-e96972c93)
General Purpose Registers:
         FP:                     0000001549ac3d00
         LR:                     00000070462044d4
         SP:                     0000001549ac3d00
         PC:                     0000007046206a3c
Start Address:                   0000007046200000
Stack Trace:
        ReturnAddress[00]:       00000070462023cc
        ReturnAddress[01]:       00000070462002cc
        ReturnAddress[02]:       0000007046200200
Stack Dump:                               00 01 02 03 04 05 06 07 08 09 0a 0b 0c 0d 0e 0f
                             001549ac3d00 70 3d ac 49 15 00 00 00 cc 23 20 46 70 00 00 00
                             001549ac3d10 01 f6 00 00 00 00 00 00 cc 02 20 46 70 00 00 00
                             001549ac3d20 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
                             001549ac3d30 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
                             001549ac3d40 00 00 00 00 00 00 00 00 30 84 21 46 70 00 00 00
                             001549ac3d50 26 00 00 00 00 00 00 00 e8 76 22 46 01 f6 00 00
                             001549ac3d60 d0 3d ac 49 15 00 00 00 f5 83 06 00 00 00 00 00
                             001549ac3d70 e0 3f ac 49 15 00 00 00 cc 02 20 46 70 00 00 00
                             001549ac3d80 58 ae 22 46 70 00 00 00 00 00 00 00 00 00 00 00
                             001549ac3d90 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
                             001549ac3da0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
                             001549ac3db0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
                             001549ac3dc0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
                             001549ac3dd0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
                             001549ac3de0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
                             001549ac3df0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
TLS Address:                 00000067f051d200
TLS Dump:                                 00 01 02 03 04 05 06 07 08 09 0a 0b 0c 0d 0e 0f
                             0067f051d200 04 00 10 00 0c 00 00 80 01 00 00 00 f5 83 06 00
                             0067f051d210 53 46 43 4f 50 02 00 00 90 3d ac 49 04 00 00 50
                             0067f051d220 53 46 43 49 00 00 00 00 02 00 00 00 00 00 00 00
                             0067f051d230 01 f6 00 00 02 00 00 00 00 00 00 00 00 00 00 00
                             0067f051d240 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
                             0067f051d250 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
                             0067f051d260 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
                             0067f051d270 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
                             0067f051d280 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
                             0067f051d290 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
                             0067f051d2a0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
                             0067f051d2b0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
                             0067f051d2c0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
                             0067f051d2d0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
                             0067f051d2e0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
                             0067f051d2f0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00

Thanks Owen.

masagrator, 2022-07-28:
Delete folder 0100000000000081 from atmosphere/contents

Mexyican, 2022-07-28:
Thank you, i will try it now and i will use the template next time.

closed 2022-12-23 by SciresM