Atmosphere-NX/Atmosphere #1928: Atmosphere don't detect Apps

Gormagala, 2022-08-01:
## Bug Report

[ If any section does not apply, replace its contents with "N/A". ]</br>
[ Lines between [ ] (square brackets) should be removed before posting. ]</br>

[ * ]</br>
[ Note: If the bug or crash you encountered is related to; ]</br>
[ - software used to make "backups", ]</br>
[ - software explicitly distributed for piracy, etc ]</br>
[ then contributors will not provide support for your issue and your issue will be closed. ]</br>

### What's the issue you encountered?

[ Describe the issue in detail and what you were doing beforehand. ]</br>
[ Did you make any changes related to Atmosphère itself? ]</br>
[ If so, make sure to include details relating to what exactly you changed. ]</br>

### How can the issue be reproduced?

[ * ]</br>
[ Include a detailed step by step process for recreating your issue. ]</br>

### Crash Report

[ Crash reports can be found under ``/atmosphere/crash_reports``. ]</br>
[ If your issue caused Atmosphère to crash, include the crash report(s) by creating a [gist](https://gist.github.com/) and pasting the link here. ]</br>
[ If you don't include a crash report in instances of crash related issues, we will ask you one to provide one. ]</br>

### System Firmware Version

1.3.2</br>

### Environment?

Atmosphère launched by: hekate
 

Error message : No applications could be found on the SD Card
Make sure a folder named /switch exists in the root of the SD Card and it contains applications.

HELP !

SciresM, 2022-08-01:
This is not a bug, and this issue tracker is not for support. Ask for help on discord.

Gormagala, 2022-08-01:
ok

Gormagala, 2022-08-01:
where i can find the discord ?

SciresM, 2022-08-01:
Do not re-open issues that are not issues. You can find the nintendo homebrew discord or the reswitched discord via google.

closed 2022-08-01 by SciresM