Atmosphere-NX/Atmosphere #1929: Separate folders for layeredfs on emunand and sysnand

DzXAnt22, 2022-08-06:
## Feature Request

The ability to have separate layeredfs folders for emunand vs sysnand. This would help so that people are able to use mods that are NOT online safe on their emunand and still be able to easily switch back to sysnand to play online again without the mod. 

### What feature are you suggesting?
#### Overview:
self explanatory 

#### Smaller Details: 
N/A

#### Nature of Request:  
[ Remove all that do not apply to your request. ]
- Addition
  - [ Ex: Addition of certain original features or features from other community projects. ]
  - [ If you are suggesting porting features or including features from other projects, include what license they are distributed under and what, if any libraries those project use. ]
- Change
- Removal
  - [Ex: Removal of certain features or implementation due to a specific issue/bug or because of low quality code, etc.]

### What component do you feel this would best fit within?
- [**Thermosphère**](https://github.com/Atmosphere-NX/Atmosphere#components)</br>
  - Atmosphère's emuNAND implementation.</br>

[ Note: **Bolded components are not implemented** or are still at the prototyping phase. ]

### Why would this feature be useful?

This feature would be useful because it lets people use mods that break online functionality or that are not online safe only on emunand offline. This removes the hassle of going back into the atmosphere folder to remove the layeredfs mods before returning online again. Big hassle for someone like me that has to constantly swap folders just to play online on that game again. (Mario kart 8 custom tracks is a good example).

SciresM, 2022-08-06:
So, sidestepping the folder separation thing because there have been many issues with that for years,

Are you aware you can hold L when launching the game to launch it without mods? This is configurable, as well.

DzXAnt22, 2022-08-06:
> So, sidestepping the folder separation thing because there have been many issues with that for years,
> 
> Are you aware you can hold L when launching the game to launch it without mods? This is configurable, as well.

woah, this is news to me actually.

The issue for me is that I don't just run a completely clean version of the game online, I just have some online safe retextures, music replaced, etc. 

can you elaborate more or link me to a guide on how to "configure" it? This might actually be exactly what I'm looking for.

Thanks!

Masamune3210, 2022-08-06:
They mean that the button to hold to disable the mods is configurable I'm pretty sure, not the actual modding itself

DzXAnt22, 2022-08-06:
> They mean that the button to hold to disable the mods is configurable I'm pretty sure, not the actual modding itself

awh man that's a shame. I could only dream of having 2 separate mod folders for each nand

closed 2022-12-23 by SciresM