Atmosphere-NX/Atmosphere #1931: Crash on Boot Error Code 2002-3502

gamer6321, 2022-08-15:
## Bug Report

### What's the issue you encountered?
I updated the switch to 14.1.2 from 13.0 and upon booting back to atmosphere received an error message
Atmosphere was left alone and not updated prior to updating FW due to being at 1.3.2
Error Code 2002-3502 (0x1b5c02)
Program 010000000000000C

### How can the issue be reproduced?
Issue can be reproduced just by rebooting the system into CFW Emunand as it goes into an endless loop of reboot and crash

### Crash Report

(https://gist.github.com/gamer6321/affe84cf160c1198ec44cdc7ec54f063#file-01660698284_010000000000000c-log)

### System Firmware Version

14.1.2</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: hekate and Fusee
- Official release or unofficial build:  Official

- Do you have additional kips or sysmodules you're loading: Not sure
- Homebrew software installed: Retroarch, MissionControl, Tinfoil, CHeckpoint
- EmuMMC or SysNAND: EmuMMC, Partition Based
  

### Additional context?

- Additional info about your environment:None

Masamune3210, 2022-08-15:
Going to hedge a guess and say your emunand is damaged and needs to be recreated, 2002-3502 is emmc read/write issues

gamer6321, 2022-08-15:
> Going to hedge a guess and say your emunand is damaged and needs to be recreated, 2002-3502 is emmc read/write issues

Would it be a clean reset or can I keep my data? if I do copy would it carry the issue over?

xkoeckiiej, 2022-10-25:
Im curious as well.
I updated Deepsea from 4.2 to 4.3 and I am getting the same error.

closed 2022-12-23 by SciresM