Atmosphere-NX/Atmosphere #1934: Panic occured while running Atmosphere.

UntitledOutput, 2022-08-24:
## Bug Report

### What's the issue you encountered?

I was just trying to get the Splatoon 3: Splatfest World Premiere by using a New Zealand account, but then my switch suddenly came up with a atmosphere error.

### How can the issue be reproduced?

I have no idea how the issue was produced, so i can't explain how it can be reproduced

### Crash Report

I don't have one. On the error screen it said it would make an error report in the fatal-errors folder, but there's just one error which I think is unrelated.

### System Firmware Version

14.1.2

### Environment?

- What bootloader  was Atmosphère launched by: hekate/fusee
- Official release or unofficial build: 1.3.2
- Do you have additional kips or sysmodules you're loading: I do not remember.
- Homebrew software installed: Atmosphere? (There was no explanation for this)
- EmuMMC or SysNAND: EmuMMC (partition)

### Additional context?
 
N/A

SciresM, 2022-08-24:
I cannot help you without the report file. The fatal error message gives you a full file path. Upload it, or this will have to be closed.

UntitledOutput, 2022-08-24:
Okay.

UntitledOutput, 2022-08-24:
[report_00000b59fd1dd6a8.zip](https://github.com/Atmosphere-NX/Atmosphere/files/9417990/report_00000b59fd1dd6a8.zip)

UntitledOutput, 2022-08-24:
Rename it to a .bin file because I couldn't upload it.

SciresM, 2022-08-24:
Post a screenshot of your atmosphere/contents folder? The error is out of range in layered romfs read, which suggests you have a broken mod of some kind.

UntitledOutput, 2022-08-24:
[image]

SciresM, 2022-08-24:
...that looks like custom sysmodule + mods, yes.

Does the error go away if you delete your mods/custom sysmodules?

For testing, you can rename the contents folder to "contents.old" to ensure clean status.

If this fixes it (and I expect it will), one of your mods is broken, you'll have to test to determine which.

UntitledOutput, 2022-08-24:
It still comes up with the error.

SciresM, 2022-08-24:
I do not actually believe you. Show me a screenshot of your atmosphere directory with no contents dir whatsoever and upload the new report.

SciresM, 2022-08-24:
(the relevant codepath in your report is impossible to trigger without a contents dir)

UntitledOutput, 2022-08-24:
I just went back to hekate, and did the usb SD card mount, and for some reason if I change the SD card (rename a file. etc.) then it reverts the changes.

SciresM, 2022-08-24:
Check that your sd card isn't write locked, I guess. Or it might be an sd card failure otherwise...I'm going to close this because it seems obvious to me this is an issue with your stuff and not an atmosphere bug, and the issue tracker is for bugs, not for support.

You may get further support on discord ^^

closed 2022-08-24 by SciresM