Atmosphere-NX/Atmosphere #1935: 0100000000000001 Breath of the Wild Fatal Error

jayvenem, 2022-08-25:
### What's the issue you encountered?

A fatal error occurred when running Atmosphère
Program ID: 0100000000000001
Error Desc: std::abort () called (0xffe)
Report saved to /atmosphere/fatal errors/report_0000000029fb71b1.bin
Press POWER to reboot

 Whenever I launch botw, i get an fatal error 0100000000000001. </br>

### How can the issue be reproduced?

I just launched breath of the wild.</br>

### Crash Report

[ https://gist.github.com/jayvenem/d7140ee9bb1d5b9862cd3956bfb0edcc ]</br>

SciresM, 2022-08-25:
Your crash report is completely unrelated to the report you're making.

Pleasae upload `atmosphere/fatal errors/report_0000000029fb71b1.bin` as mentioned in your issue.

jayvenem, 2022-08-25:
> Your crash report is completely unrelated to the report you're making.
> 
> Pleasae upload `atmosphere/fatal errors/report_0000000029fb71b1.bin` as mentioned in your issue.

Apologies!

jayvenem, 2022-08-25:
[report_0000000029fb71b1.zip](https://github.com/Atmosphere-NX/Atmosphere/files/9420599/report_0000000029fb71b1.zip)

jayvenem, 2022-08-25:
It wouldn't allow me to send the bin file.

SciresM, 2022-08-25:
Out of bounds error when applying an ips patch.

I think your game mod is corrupted, please delete it + verify issue is fixed, then re-download.

This isn't a bug, and the issue tracker is for bugs (not support) so closing -- further support is available on discord as needed.

SciresM, 2022-08-25:
(Your mod will presumably be at atmosphere/contents/01007ef00011e000)

closed 2022-08-25 by SciresM

jayvenem, 2022-08-25:
> (Your mod will presumably be at atmosphere/contents/01007ef00011e000)

Nothing was there oddly, reinstalled and just crashed again, strange.

SciresM, 2022-08-25:
Post a screenshot of your atmosphere/contents folder?

jayvenem, 2022-08-25:
[image]
[image]