Atmosphere-NX/Atmosphere #1938: LayeredFS replace entire directory suppressing existing contents

fennectech, 2022-09-06:
## Feature Request

### What feature are you suggesting?
#### Overview:
- Replace entire directory contents with contents on SD 
- 
#### Smaller Details: 
- i would like a way that i can replace the entire contents of a directory instead of just replacing / adding individual files in it.  So that files originaly there are not seen by the game
- 
#### Nature of Request:  
- Addition
  - Custom system modules.</br>
  - Application-level patches to the Horizon OS.</br>

### Why would this feature be useful?
This would be useful to replace entire directories in games with different contents and completely suppress the existing files.

fennectech, 2022-09-06:
Because its simper    There are some games that load all contents in a spesific directory and being able to suppress files without replacing them would be great

SciresM, 2022-09-06:
I don't actually agree or believe that it's simpler.

fennectech, 2022-09-06:
It’s easier to pull off for those with limited programming abilities.  It also is less maintaince for games with frequent updates.Sent from my iPad

closed 2022-12-23 by SciresM