Atmosphere-NX/Atmosphere #1939: I cant build It

fengahan, 2022-09-06:
Ubuntu 22.04
Python 3.x

make[4]: “/home/desmond/下载/Atmosphere/fusee/program/out/nintendo_nx_arm_armv4t/release/program.lz4”已是最新。
Checked program.
Checking loader stub...
Checked library.
Checked fusee program.
fusee_loader_main.cpp
/home/desmond/下载/Atmosphere/fusee/loader_stub/source/fusee_loader_main.cpp:18:10: fatal error: program_lz4.h: No such file or directory
   18 | #include "program_lz4.h"
      |          ^~~~~~~~~~~~~~~
compilation terminated.
make[4]: *** [/opt/devkitpro/devkitARM/base_rules:80：fusee_loader_main.o] 错误 1
make[3]: *** [/home/desmond/下载/Atmosphere/fusee/loader_stub/loader_stub.mk:54：all] 错误 2
make[2]: *** [/home/desmond/下载/Atmosphere/fusee/fusee.mk:25：check_loader_stub] 错误 2
make[1]: *** [/home/desmond/下载/Atmosphere/atmosphere.mk:121：fusee] 错误 2
make: *** [/home/desmond/下载/Atmosphere/Makefile:39：nx_release] 错误 2
make[1]: 离开目录“/home/desmond/下载/Atmosphere”
make: 离开目录“/home/desmond/下载/Atmosphere”

Process finished with exit code 2

SciresM, 2022-09-06:
Closing because this kind of thing is almost certainly an environment issue...program_lz4.h should exist if fusee program has been checked.

I would try both clean and just making again.

Support is available on discord for this kind of thing.

closed 2022-09-06 by SciresM

fengahan, 2022-09-08:
Thank you very much, I don't know how I can get program_lz4.h

fengahan, 2022-09-08:
I have tried clearAll many times, but it still doesn't work