Atmosphere-NX/Atmosphere #1940: Error on starting

ELNuggetin, 2022-09-10:
## Bug Report

[ If any section does not apply, replace its contents with "N/A". ]</br>
[ Lines between [ ] (square brackets) should be removed before posting. ]</br>

[ * ]</br>
[ Note: If the bug or crash you encountered is related to; ]</br>
[ - software used to make "backups", ]</br>
[ - software explicitly distributed for piracy, etc ]</br>
[ then contributors will not provide support for your issue and your issue will be closed. ]</br>

### What's the issue you encountered?

I'm trying to open atmosphere with Hekate, at the moment I execute atmosphere, it shows the loading screen, and then the Nintendo's loading screen. It turns into a black screen.

### How can the issue be reproduced?

I'm just opening atmosphere commonly.

### Crash Report

Þ ¸SystemPhysicalMemoryPeakÓ    ýÀ µSystemThreadCountPeakÓ      )´SystemEventCountPeakÓ      ½SystemTransferMemoryCountPeakÓ       ¶SystemSessionCountPeakÓ      ‚¹SystemPhysicalMemoryLimitÓ    0  ¶SystemThreadCountLimitÓ      `µSystemEventCountLimitÓ      „¾SystemTransferMemoryCountLimitÓ       ˆ·SystemSessionCountLimitÓ      g©AbortFlagÂ­HasSyslogFlagÃ©OsVersion¦14.1.2°PrivateOsVersionÙ1NintendoSDK Firmware for NX 14.1.2-1.0 (ad52f498)¬SerialNumber®XAW00000000000°ReportIdentifierÙ$642eda95-2e3c-4f41-4de4-6bfa2167e613³OccurrenceTimestampÓ    cÇl¶OccurrenceTimestampNetÓ    cÇl´ReportVisibilityFlagÂ®OccurrenceTickÓ    îbs¹SteadyClockInternalOffsetÓ        Ù SteadyClockCurrentTimePointValueÓ    	Š·ElapsedTimeSincePowerOnÓ        ¹ElapsedTimeSinceLastAwakeÓ        ­RegionSetting¦Europe¬ProductModel¢NX©ErrorCode©2162-0004©CipherKeyÅ                                                                                                                                                                                    
### System Firmware Version

14.1.2

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: Hekate
- Official release or unofficial build: 

- EmuMMC or SysNAND: EmuMMC
  -Partition

### Additional context?

- Additional info about your environment:
Idk

Masamune3210, 2022-09-10:
upload the bin please, not open it and splat its contents into the issue

closed 2022-10-12 by SciresM

HunterBoy344, 2024-09-14:
Also having this issue. What was the solution?