Atmosphere-NX/Atmosphere #1942: Nintendo switch won't boot into stock or emummc or sysmmc or fuze i think i bricked it need help

Needhelpforswitch, 2022-09-21:
## Bug Report

[ If any section does not apply, replace its contents with "N/A". ]</br>
[ Lines between [ ] (square brackets) should be removed before posting. ]</br>

[ * ]</br>
[ Note: If the bug or crash you encountered is related to; ]</br>
[ - software used to make "backups", ]</br>
[ - software explicitly distributed for piracy, etc ]</br>
[ then contributors will not provide support for your issue and your issue will be closed. ]</br>

### What's the issue you encountered?

[Hello i need help for my switch it doesn't boot when i boot into anything it shows up for 1 second then shuts down.

It happend like this i was just ready to play a game and injected the hekate payload into my switch threw my phone and the console booted up into hekate i launched emummc and it booted normally it showed my custom theme i installed thru nxthemes i launched the game and then went back into the homemenu but then when i went back it showed a giant red battery icon and shut down my console (i have and had auto rcm enabled) so i tried to boot back into hekate and since then it doesn't boot into anything (sysmmc or emummc or stock or fuze) i have my sd card on fat32 and i've seen soo many post and tried a lot of different types of fixes and even restored my old nand (1 month old) and the sysnand or stock still didn't boot does someone have a fix for it cause im out of ideas.

Edit 1:Btw since the nintendo switch crashed with auto rcm on the battery is really low and the readings are a little weird could that prevent it from booting into any firmware?. ]</br>
[  ]</br>
[ If so, make sure to include details relating to what exactly you changed. ]</br>

### How can the issue be reproduced?

[ No idea ]</br>

### Crash Report

No crash reports found on the sd card

### System Firmware Version

14.1.2 i think</br>

SciresM, 2022-09-21:
The atmosphere issue tracker is for bugs. You're asking for support, and just describing the low battery icon that shows up when you need to charge your console before turning it on.

Seek further support on discord.

closed 2022-09-21 by SciresM

Needhelpforswitch, 2022-09-21:
i tried discord but the link doesn't allow me to join discord don't know why