Atmosphere-NX/Atmosphere #1945: Fatal error on boot

MitchMakesThings, 2022-10-08:
### What's the issue you encountered?

Booting from Hekate (5.8.0) to emummc firmware.
Atmosphere logo shows briefly, then a black screen with the charging icon in the top right.
The screen goes black for a few seconds, then displays the error message:
A fatal error occurred when running Atmosphere.
Program Id: 010000000000002b
Error Desc: std::abort() called (0xffe)

### How can the issue be reproduced?

Booting to custom firmware on emummc

### Crash Report

Will upload shortly

### System Firmware Version
Unsure, can't boot to the firmware. I don't think I've updated in the last year though.

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: Hekate
- Official release or unofficial build: Official 1.2.6
- Do you have additional kips or sysmodules you're loading: No
- Homebrew software installed: None
- EmuMMC or SysNAND:
  - EmuMMC, not sure if partition or file based

### Additional context?

I was trying to update Atmosphere after a long time being away. The latest release didn't work, so I've tried 1.2.6. Same result both times.

MitchMakesThings, 2022-10-08:
[report_000000000ae797a0.zip](https://github.com/Atmosphere-NX/Atmosphere/files/9740455/report_000000000ae797a0.zip)

Here's the crash report from 1.2.6

SciresM, 2022-10-12:
You have a corrupt PRODINFO/system save.

Follow the instructions/directions in https://github.com/Atmosphere-NX/Atmosphere/issues/1256 for guidance. You'll need to restore a non-corrupt backup and/or delete the system settings savedata.

closed 2022-10-12 by SciresM