Atmosphere-NX/Atmosphere #1946: 15.0.0 support

SciresM, 2022-10-11:
Master issue/check list of stuff to do for 15.0.0 changes.

- [ ] Reverse engineer new kernel changes 
- [ ] Determine what system modules need reverse engineering changes
- [ ] Reverse engineer new erpt changes
- [ ] Reverse engineer new pm changes
- [ ] Reverse engineer new fatal changes
- [ ] Update fusee/exosphere with new keydata/values
- [ ] Implement new changes into mesosphere
- [ ] Implement new changes into erpt
- [ ] Implement new changes into pm
- [ ] Implement new changes into fatal
- [ ] Update emummc
- [ ] Miscellaneous exosphere/fusee fixes for new DRAM IDs
- [ ] Make sure libnx is updated for anything relevant.

Post-release:
- [ ] Reverse engineer potential new boot sysmodule changes.
- [ ] Reverse engineer loader sysmodule changes.
- [ ] Miscellaneous post-release reverse engineering
- [ ] Miscellaneous post-release implementation work/updating

jvan1234, 2022-10-11:
Did update, and got a package1 error. Thank you for providing a checklist!

SciresM, 2022-12-23:
Did I just never close this? Man.

closed 2022-12-23 by SciresM