Atmosphere-NX/Atmosphere #1948: The software was closed because an error occurred when playing Star Wars: Knights of the Old Republic 2

NaturalHarmonia, 2022-10-11:
## Bug Report

### What's the issue you encountered?

[ While playing Star Wars: Knights of the Old Republic 2 I occasionally get 'The software was closed because an error occurred. ]</br>
[ I haven't recently made any changes to Atmosphere itself. ]

### How can the issue be reproduced?

</br>
[ The issue appears randomly while playing the game. There doesn't seem to be much rhyme or reason to it. ]</br>

### Crash Report

[ (https://gist.github.com/NaturalHarmonia/11aedbcf66f1364bab82c44d630052ce) ]</br>

### System Firmware Version

12.1.2</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: fusee
- Official release or unofficial build: 
  - [ Official release version AMS 1.3.2 ]
  - Do you have additional kips or sysmodules you're loading: N/A
- Homebrew software installed: [ Breeze, All-in-One Switch Updater, Daybreak, DBI, EdiZon, EdiZon SE, Goldleaf, JKSV, Reboot to Payload, and RetroArch ]
- EmuMMC or SysNAND: SysNAND

### Additional context?

- Additional info about your environment: N/A

sudot0ph, 2022-10-11:
Try to hold L while launching the game and see if issue still persists. If so, there are cheats activated that break the game. Either delete cheats or customize system_settings.ini to disable cheats by default.

NaturalHarmonia, 2022-10-11:
> Try to hold L while launching the game and see if issue still persists. If not, there are cheats activated that break the game. Either delete cheats or customize system_settings.ini to disable cheats by default.

Where are cheats usually located?

sudot0ph, 2022-10-11:
/atmosphere/content/TID of game in question

You can use AiO Updater to delete cheats completely or for certain titles.

Masamune3210, 2022-10-11:
Piracy is not supported.

NaturalHarmonia, 2022-10-11:
> /atmosphere/content/TID of game in question
> 
> You can use AiO Updater to delete cheats completely or for certain titles.

I don't have a content folder so that means I shouldn't have any cheats at all in the first place.

SciresM, 2022-10-12:
This sounds like a game issue, tentatively -- the thread resource limit is being hit. I just double checked, and the runtime limit on application reslimit threads is correct...maybe the game expects the boosted limit? But I don't think that's done on prod...

closed 2022-10-12 by SciresM