Atmosphere-NX/Atmosphere #1953: nosigchk error, if continue no games launch

sidnegel, 2022-10-12:
## Bug Report

[ nosigchk upon launch]</br>
[ when launching any application, skynx, uno, pokemon, etc.. it says that an error has occurred and sends me to the menu, if i try to launch the application again it says the software is corrput, if i launch it a third time it goes back to saying an error has occurred and then back to corrupt and so on and so on.]</br>

### What's the issue you encountered?

[after using prerelease 1.4 i get an error stating nosigchk, im given two options 1. press power to continue and 2. press vol- to return to menu in hekate 5.9.0, was previously using 1.3.2 atmos and 5.8.0 hekate.  i continue past the error and it boots into atmos on 15.0 but no applications run even with gamecard inserted, i receive an error message after selecting my profile and im prompted to return to the home screen.  after returning to the homescreen if i select the application again it says its corrupted.  this process repeats for every application, even ones like skynx.  effectively locking me in applet mode in the hombrew menu.]</br>
[i updated to 15.0 via daybreak like a dumbass while on 1.3.2 atmos]</br>
[see above]</br>

### How can the issue be reproduced?

[issue is constant, restarts do nothing]</br>
[see first paragraph]</br>

### Crash Report

[N/A]</br>

### System Firmware Version
15.0.0</br>

### Environment?

- What bootloader Atmosphère launched by: Hekate 5.9.0 official release

- Do you have additional kips or sysmodules you're loading:
- Homebrew software installed: [ * ]
- EmuMMC or SysNAND:
  - [partition based EmuMMC]

### Additional context?

- Additional info about your environment:
- [your guess is as good as mine, hopefully i dont have to format lol]

SciresM, 2022-10-12:
Piracy is not supported.

closed 2022-10-12 by SciresM

sidnegel, 2022-10-12:
what does this have to do with piracy?

SciresM, 2022-10-12:
"nosigchk" = piracy patches you have enabled. 

You then describe a bunch of pirate-installed content that isn't launching because it's pirated.