Atmosphere-NX/Atmosphere #1954: Compiling error

impeeza, 2022-10-13:
## Bug Report

### What's the issue you encountered?

I am trying to compile the latest code (e580c39) using MSYS MINGW64 but get an error on libstratosphere section

```Checked boot library.
ams::Result ams::svc::aarch64::lp64::MapInsecu                                                                                                                             ureMemory(ams::svc::Address, ams::svc::Size)':
Checked library.
C:/msys64/home/Impeeza/Atmosphere/libraries/libstratosphere/include/stratosphere/svc/svc_stratosphere_shims.hpp:507:32: error: '::svcMapInsecureMemory' has not been declared; did you mean 'MapInsecureMemory'?
  507 |                     R_RETURN(::svcMapInsecureMemory(reinterpret_cast<void *>(static_cast<uintptr_t>(address)), size));
      |                                ^~~~~~~~~~~~~~~~~~~~
c:\msys64\home\impeeza\atmosphere\libraries\libvapours\include\vapours\results\results_common.hpp:436:48: note: in definition of macro 'R_RETURN'
  436 |         const ::ams::Result _tmp_r_throw_rc = (res_expr);                                                                                      \
      |                                                ^~~~~~~~
C:/msys64/home/Impeeza/Atmosphere/libraries/libstratosphere/include/stratosphere/svc/svc_stratosphere_shims.hpp: In function 'ams::Result ams::svc::aarch64::lp64::UnmapInsecureMemory(ams::svc::Address, ams::svc::Size)':
C:/msys64/home/Impeeza/Atmosphere/libraries/libstratosphere/include/stratosphere/svc/svc_stratosphere_shims.hpp:511:32: error: '::svcUnmapInsecureMemoryChecked library.
' has not been decla                                                                                                                                                       ared; did you mean 'UnmapInsecureMemoryChecking kernel...
Checking kernel ldr...
'?
  511 |                     R_RETURN(::svcUnmapInsecureMemorymake[4]: '/home/Impeeza/Atmosphere/libraries/libexosphere/lib/nintendo_nx_arm_armv4t/release/libexosphere.a' is up to date.
(reinterpret_cast<void *>(static_cast<uintptr_t>(address)), size));
      |                                warmboot_bootrom_workaround.cpp
warmboot_clkrst.cpp
^~~~~~~~~~~~~~~~~~~~~~warmboot_cpu_cluster.cpp
Checked library.
warmboot_dram.cpp

warmboot_main.cpp```

![image](https://user-images.githubusercontent.com/41979604/195479511-748dd3bc-3cfb-4381-9ba0-28d707e155d2.png)

the final error is:

built ... kernel.bin
Checked kernel.
Built mesosphere.bin...
make[1]: Leaving directory '/home/Impeeza/Atmosphere'
make: *** [Makefile:39: nx_release] Error 2

on exactly the same MSYS a copy of  atmosphere 1.3.2 master 28f0940c5 downloaded on september 16th compile just fine.

There is any can I do?  

thanks.

### How can the issue be reproduced?
Try to compile using MSYS

### Crash Report
N/A

### System Firmware Version
N/A

### Environment?
N/A

### Additional context?
N/A

SciresM, 2022-10-13:
AMS requires the currently-unmerged branch of libnx that adds 15.0.0 support. You can either install that locally or wait a few days.

closed 2022-10-13 by SciresM

impeeza, 2022-10-13:
Thanks a lot. will investigate how to make it,   regards from Colombia.