Atmosphere-NX/Atmosphere #1955: 15.0.0 unable to start software

Killua0017, 2022-10-13:
### What's the issue you encountered?
I accidentally update my software from 14.2 to 15.0.0 the latest software.
I replace my atmosphere to latest one (pre-release version)
I successfully replace the atmosphere to my sd card with new version and inject the new fusee.bin.
my console is starting but all software inside is not working it says
Unable to start software.
Return to the HOME Menu and try again.
 

### System Firmware Version

current version : 15.0.0/AMS 1.4.0S

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: FUSEE
- Official release or unofficial build: 
  - [ Official release version x.x.x (or) unofficial build ] Pre release
  - [ If using an unofficial build, include details on where/how you acquired the build. ] 
     - [ Ex: Self-compilation ]
     - [ Ex: Kosmos' distribution of Atmosphère ]
- Do you have additional kips or sysmodules you're loading:  I dont know
- Homebrew software installed: [ * ]
- EmuMMC or SysNAND:
  - [ If using an EmuMMC, include whether it's partition-based or file-based. ] file based i think

### Additional context?

- Additional info about your environment:
- [ Any other information relevant to your issue. ]
[image]
[image]
[image]

JustNightBG, 2022-10-13:
Did you tried updating the sigpatches?
https://jits.cc/patches

Killua0017, 2022-10-13:
Thank you very much.
i tried this one . now its working but only tinfoil is not working. when i open the tinfoil it stuck on black screen. but other saved software is working

 
 

https://jits.cc/patches

masagrator, 2022-10-13:
Piracy is not supported

closed 2022-10-13 by Killua0017