Atmosphere-NX/Atmosphere #1956: strange comment on commit

impeeza, 2022-10-14:
Hi, it is not a bug, but the comment on the commit (https://github.com/Atmosphere-NX/Atmosphere/commit/77fe5cf6f574a0df4198015af672acfad4b88acb) have a strange comment:

[image]

```
 ams: the copyright^H^H^H^H^H^H^Hmplex plane is the algebraic closure …
…of the reals
```

could be just something decorative but could not.

regards.

DavidBuchanan314, 2022-10-14:
It's a meme

SciresM, 2022-10-14:
That commit is a shitpost, yeah. The source phrase is `the complex plane is the algebraic closure of the reals`; the entirety of the humor is that "copyright" shares the first two letters with "complex".

`^H` is what a terminal which poorly parses user input will show when a backspace is input.

I think it's pretty hilarious. There are some other commits in atmosphere like that: https://github.com/atmosphere-NX/Atmosphere/commit/96937a611dffa9a04e4d41f4d972ecd9bcc8aa70, https://github.com/atmosphere-NX/Atmosphere/commit/5a8aebf74d92cd45f238118904ccedd64ff77a26, https://github.com/atmosphere-NX/Atmosphere/commit/931e38742beac3ad1acf05cdb4855e93d40947eb

Don't worry about dumb commit messages. I like to make myself laugh.

closed 2022-10-14 by SciresM

impeeza, 2022-10-14:
😂 humor is important.  Thanks for all the effort