Atmosphere-NX/Atmosphere #1957: Can't boot into CFW or OFW

Alvyssy, 2022-10-14:
## Bug Report

[ If any section does not apply, replace its contents with "N/A". ]</br>
[ Lines between [ ] (square brackets) should be removed before posting. ]</br>

[ * ]</br>
[ Note: If the bug or crash you encountered is related to; ]</br>
[ - software used to make "backups", ]</br>
[ - software explicitly distributed for piracy, etc ]</br>
[ then contributors will not provide support for your issue and your issue will be closed. ]</br>

### What's the issue you encountered?

This morning I woke my switch up from sleep and everything was fine, but when I tried putting it to sleep it displayed an error code that I failed to collect because I thought it was fine, rebooted to hekate and tried rebooting into atmosphere, all fine again but putting the console to sleep and it crashed again. Tried rebooting again and now the console doesn't boot at all.

Trying to boot into OFW gets stuck on a black screen  and trying to boot CFW gets stuck on the Nintendo Switch logo after the atmosphere logo.

My nintendo switch is OLED and I have the latest pre release of atmosphere, I tried putting the 1.3.2 version and it gave me a blue screen obviously

My problem is that ofw and cfw via hekate boots normally in sleep mode and now I can't boot my switch anymore, is there anything I can do to fix?

### How can the issue be reproduced?

Turn on the console 
Try to boot either ofw or cfw
Put it to sleep

### Crash Report

I don't have a crash_reports folder

### System Firmware Version

15.0.0

### Environment?

- Hekate via HWFLY modchip
- Official release or unofficial build:  1.4.0 Atmosphere pre-release that I got here
- Do you have additional kips or sysmodules you're loading:  sys-clk and tesla menu
- No homebrew software installed 
- SysNAND

### Additional context?

- Additional info about your environment:
- [ Any other information relevant to your issue. ]

masagrator, 2022-10-14:
If this is not Atmosphere exclusive related problem, better come into ReSwitched discord and ask there. Issues is not a place for support but for bug reports related Atmosphere.

Alvyssy, 2022-10-14:
Sorry I will follow your advice

closed 2022-10-14 by SciresM