Atmosphere-NX/Atmosphere #1959: Bluetooth controllers dont work

PistonTomas, 2022-10-14:
## Bug Report

[ If any section does not apply, replace its contents with "N/A". ]</br>
[ Lines between [ ] (square brackets) should be removed before posting. ]</br>

[ * ]</br>
[ Note: If the bug or crash you encountered is related to; ]</br>
[ - software used to make "backups", ]</br>
[ - software explicitly distributed for piracy, etc ]</br>
[ then contributors will not provide support for your issue and your issue will be closed. ]</br>

### What's the issue you encountered?

I updated to 15.0.0 and atmosphére 1.4.0 a when i want to connect my pro controller using bluetooth didn connects

### How can the issue be reproduced?

Connecting a bluetooth controller

### Crash Report

### System Firmware Version
15.0.0 and atmosphére 1.4.0 
### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: Hekate
- Official release or unofficial build:  Official build 1.4.0
  - [ Official release version x.x.x (or) unofficial build ]
  - [ If using an unofficial build, include details on where/how you acquired the build. ] 
     - [ Ex: Self-compilation ]
     - [ Ex: Kosmos' distribution of Atmosphère ]
- Do you have additional kips or sysmodules you're loading: sys-con ,sys-clk
- Homebrew software installed: [ All SDSetup  ]
- EmuMMC or SysNAND: SysNAND

### Additional context?

SciresM, 2022-10-14:
I would bet this is to do with your custom sysmodule that impacts controller connections.

Atmosphere doesn't touch hid/Bluetooth stuff.

closed 2022-10-14 by SciresM

Masamune3210, 2022-10-14:
Make sure you update Mission Control if you are using it, it recently got a tentative update to support 15.0.0