Atmosphere-NX/Atmosphere #1962: Unable to launch game

Ramon33620, 2022-10-15:
## Bug Report

I updated atmosphere this morning for Atmosphere 1.4.0
but now all games doesn't work anymore... without message, bug or crash
It's like official switch : "sofware can't be started"

### What's the issue you encountered?
I updated atmosphere this morning for Atmosphere 1.4.0
but now all games doesn't work anymore... without message, bug or crash
"sofware can't be started" like official switch

### How can the issue be reproduced?
with all games

### Crash Report
no crash 

### System Firmware Version

15.0.0
X.X.X</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: fusee, TegraRcmGUI 2.6
- Official release or unofficial build: official
  - [ Official release version x.x.x (or) unofficial build ] 
  - [ If using an unofficial build, include details on where/how you acquired the build. ] 
     - [ Ex: Self-compilation ]
     - [ Ex: Kosmos' distribution of Atmosphère ]
- Do you have additional kips or sysmodules you're loading: not sure
- Homebrew software installed: [ * ] no
- EmuMMC or SysNAND: 
  - [ If using an EmuMMC, include whether it's partition-based or file-based. ]

### Additional context?

- Additional info about your environment:
-  it's first time I updated atmosphere, I bought my switch 1 month ago... maybe I forgot to do something

SciresM, 2022-10-15:
Re-download games from the eShop if necessary.

The issue tracker is for bugs, not for support.

closed 2022-10-15 by SciresM

Wirexuser01, 2022-10-16:
@Ramon33620 Sigpatches AMS 1.4.0

Ramon33620, 2022-10-16:
Thank you! @SciresM