Atmosphere-NX/Atmosphere #1965: A fatal error occurred when running Atmosphere. Title ID: 010041544d53000

SoulPunisher1, 2022-10-16:
## Bug Report

A fatal error occurred when running Atmosphere.
Title ID: 010041544d530000
Error Desc: std::abort() called (0xffe)
Report saved to /atmosphere/fatal_errors/report_0000000043307ffe.bin

### What's the issue you encountered?

Updated atmosphere, fusee, and hekate to the latest stable versions, booted up the hekate mod menu, pressed fusee, the atmosphere logo popped up followed by a black screen, then the error shows. Tried the process again with emmuc and still the same thing happened. I CAN however launch the switch in stock mode. 

### How can the issue be reproduced?

Boot hekate
launch fusee or emummc
atmosphere logo shows up followed by black screen
report error

### Crash Report

https://gist.github.com/SoulPunisher1/1b221147969b3658ad87d8dcfb247126 <--- fusee

https://gist.github.com/SoulPunisher1/5b3a4d4209e585be1bd13b336e44681b <--- emummc

### System Firmware Version

11.1.0

closed 2022-10-16 by SciresM

AdrienTheThird, 2022-10-16:
I had the same problem, after the upgrade to Atmosphere v1.4.0 I only got the Atmosphere logo and a black screen after that.

My Fix was to just delete my old /atmosphere/ and /sept/ folders and use the new ones from the 1.4.0 release.
After that everything worked fine again.

I don't know if that will help the devs, but maybe it at least helps @SoulPunisher1.

SciresM, 2022-10-16:
That is a different and completely unrelated problem, @AdrienTheThird.

SoulPunisher1, 2022-10-16:
Is this what you mean by re-creating an emummc?

 https://switch.homebrew.guide/emummc/emummc.html

SciresM, 2022-10-16:
That's for creating, not recreating, but sure approximately.

Please note the issue tracker is for bugs, not for support. Please seek further support on discord.