Atmosphere-NX/Atmosphere #1970: Panic occured while running atmosphere, help please

Crooxy, 2022-10-19:
## Bug Report
N/A

### What's the issue you encountered?
Recently I've tried to update Atmosphere. Before the update, everything was working fine, but now Atmosphere won't boot. 

When I'm trying to launch CFW - emuMMC, an error is detected. 
**It says:** 
Panic occurred while running Atmosphere
Title ID : 010000000000BD00
Error : std : : abort (0xFFE)

### How can the issue be reproduced?

I went on GitHub and I've downloaded the latest Atmosphere version 1.4.0

### Crash Report

Picture link to the post

### System Firmware Version

1.4.0 Atmosphere
v5.9.0 for Hekate

### Environment?

- What bootloader (fusèe, Hekate, etc) was Atmosphère launched by: **Hekate v5.9.0**

- Do you have additional kips or sysmodules you're loading: **I don't know** 
- Homebrew software installed: **yes**
- EmuMMC or SysNAND: **EmuMMC**

### Additional context?

Please if you can help me via this post or via discord Crooxy#5803 

Thanks!!

SciresM, 2022-10-19:
Not an atmosphere issue; update mission control.

closed 2022-10-19 by SciresM

Crooxy, 2022-10-19:
I've tried but for some reason, it doesn't work...

SciresM, 2022-10-19:
Delete mission control, then. Locking this as resolved.