Atmosphere-NX/Atmosphere #1972: Cannot use fs.midm on overlayDisp

Ereza, 2022-10-20:
## Bug Report

### What's the issue you encountered?

When an attempt is made to use fs.midm on `overlayDisp` (title id 010000000000100C), the system hangs at startup.

I use fs.mitm to provide a Catalan language translation to the Switch interface (https://github.com/Ereza/switch-cat). This is done by replacing some .szs files on each system title via fs.mitm. I was using firmware 13.0.0 and Atmosphère 1.1.1 until now, but now I upgraded to firmware 15.0.0 and Atmosphère 1.4.0.

When on firmware 13.0.0 and Atmosphère 1.1.1 it was possible to us fs.mitm on `overlayDisp` with no problems. However, under the current versions, adding any file to `atmosphere/contents/010000000000100C/romfs/` completely hangs the system at startup. Just a guess, but could this be related to the optimizations made to fs.mitm in 1.2.4?

### How can the issue be reproduced?

* Create the directory `atmosphere/contents/010000000000100C/romfs/` on your SD card, and place an empty `dummy.txt` file inside the directory.
* Power on the console.
* The system hangs on a black screen after the Nintendo Switch logo.

If you place no files inside the `atmosphere/contents/010000000000100C/romfs/` directory, the console boots just fine.

### Crash Report

N/A

### System Firmware Version

15.0.0

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: fusee
- Official release or unofficial build: 
  - Official, 1.4.0
- Do you have additional kips or sysmodules you're loading: No.
- Homebrew software installed: Just Daybreak, Goldleaf and nxdumptool.
- EmuMMC or SysNAND: SysNAND.

### Additional context?

- Additional info about your environment:
  - Switch console from 2017 (first model).
  - `nogc = 0`.

SciresM, 2022-10-20:
You report the issue doesn't occur on 1.1.1/13.0.0, and that it does occur on 1.4.0/15.0.0.

Does it occur on 1.4.0/13.0.0?

Ereza, 2022-10-20:
Unfortunately I did not try it on 1.4.0/13.0.0, I updated the firmware and Atmosphère at the same time...

SciresM, 2022-10-20:
Could you make an emummc, downgrade it to 13.0.0, and test it? If not, it is not the end of the world, but it may take me longer to test and reproduce locally.

SciresM, 2022-10-20:
No worries, I will investigate myself.

Ereza, 2022-10-20:
Oh, I didn't mention that, but just in case: other system titles work as expected when using fs.mitm on them. It seems to happen only with `overlayDisp`.

Also, possible related issue with what could be the same problem: #1698.

JarleSoelie, 2022-10-31:
Hi,
I believe I'm getting the same error message after updating atomsphere, hekate, and then the firmware to 15.0. Everything was working great before, and I have also updated everything several times before without any problems. I never installed any themes, and I don't have the 010000000000100C folder in my atmosphere/contents folder. 

I am using emummc, (that I have left be on the sd card since I really don't want to delete 350 gb of data), but I tried to delete absolutely everything else regarding atmosphere and hekate and do as fresh of an install as possible, but I still get the exact same error message no matter what I do... I have attached the fatal error log below. Is there something inside the emummc that can cause this? Is there anything that can be done you think, or is perhaps the SD card itself on the fritz?
[01640995215_010000000000100c.log](https://github.com/Atmosphere-NX/Atmosphere/files/9905283/01640995215_010000000000100c.log)

JarleSoelie, 2022-11-01:
> Hi, I believe I'm getting the same error message after updating atomsphere, hekate, and then the firmware to 15.0. Everything was working great before, and I have also updated everything several [...]

Wait! I fixed it, but I'm not entirely sure how. I made sure no overlays were active, I reinstalled atmosphere and hekate (making sure I extracted them from their archives before copying the files to the SD card). Then I used the latest TegraRCMgui instead of my phone to load the hekate payload. And suddenly I was in! I apologize for posting this here, I was at my wits end last night trying to figure this error out. Thanks for everything

Ereza, 2023-05-10:
I just wanted to confirm that this still happens in Atmosphère 1.5.3 with firmware 16.0.3. Creating any file inside the `romfs` directory for title id `010000000000100C` hangs the system on startup.

SciresM, 2025-04-03:
Not closing this issue of the various overlayDisp redirection stuff, but noting that I have no idea how to fix this or why the problem occurs.

commit 98bc030b (2025-09-03): ams.mitm: fix deadlock when building multiple romfs simultaneously (closes #1972)
closed 2025-09-03 by SciresM