Atmosphere-NX/Atmosphere #1973: Force enable dock usb ports while sleeping

fennectech, 2022-10-20:
## Feature Request

Id like a way to force the system to keep the USB ports awake while the console sleeps in docked mode.  This would allow periferals like the MAGIC NS and such to wake the console from sleep even without the ethernet adapter being used

### What feature are you suggesting?
#### Overview:
The switch currently allows the console to keep the USB ports awake if you connect to the internet via a USB ethernet adapter.  This enables you to wake the console via a USB controller adapter like the magic NS.  
The feature request is to be able to keep it awake without that ethernet adapter. 

#### Smaller Details: 
I dont know any more detailed

#### Nature of Request:  
[ Remove all that do not apply to your request. ]
- Addition

### What component do you feel this would best fit within?
- [Stratosphère](https://github.com/Atmosphere-NX/Atmosphere#components)</br>
    - Custom system modules.</br>
[ Note: **Bolded components are not implemented** or are still at the prototyping phase. ]

### Why would this feature be useful?
To let usb controler adapters wake the console without having an ethernet adapter   This would also allow users to access sys-ftpd while the console is asleep

SciresM, 2022-10-20:
Sounds like a good idea for a custom sysmodule! I recommend doing the research and writing one to do this, I don't think it makes sense for built in integration in atmosphere.

closed 2022-10-20 by SciresM

fennectech, 2022-10-20:
Could you link to a switchbrew article that shows it worksSent from my iPad

Closed #1973 as completed.

SciresM, 2022-10-20:
There is no such article. I don't even know how you'd do this. You'll have to do the research.

fennectech, 2022-10-20:
Found how you’d do it.   https://switchbrew.org/wiki/Network_Interface_services#SetKeptInSleep

SciresM, 2022-10-20:
Closing, since this is no longer an atmosphere problem. Good luck implementing!