Atmosphere-NX/Atmosphere #1974: hbmenu complaing there are no files?

solomani, 2022-10-21:
### What's the issue you encountered?

I decided to use a 256gb sdcard to dump games for yuzu, I was using a 32gb previously.  I just copied the files from the 32 to the 256.  All works fine except when I try and start hbmenu it complains "No applications could be found on the SD card. Make sure a folder named /switch exists in the root of the SD card, and it contains applications".

The folder does exist as it's a direct copy of the 32gb card which works fine.

### How can the issue be reproduced?
n/a

### Crash Report
n/a, no crashing just no menu options.

### System Firmware Version
14.1.2

### Environment?
hekate_ctcaer_5.9.0_Nyx_1.4.0
fusee 5.9
hbmenu v3,5,1
nx-hbloader v2.4.1
atmosphere-1.4.0-master-36a33617a+hbl-2.4.3+hbmenu-3.5.1

- Homebrew software installed: [ * ]
as above (all files required for yuzu)

### Additional context?
n/a

Thanks.

SciresM, 2022-10-21:
This is decidedly not an atmosphere problem, and the issue tracker is for bugs, not for support.

I would guess you have the archive bit set by accident so the os thinks the switch folder is a file. Try unsetting it on the switch folder, and if that doesn't work seek further support on discord.

closed 2022-10-21 by SciresM

solomani, 2022-10-21:
Thank you, what is your discord link?

SciresM, 2022-10-21:
When you hit "create issue", you're presented with "bug report", "feature request", and a link to the discord for everything else.

ReSwitched is very technical (and has a technical competency entrance test), you may also try the Nintendo Homebrew discord, link obtainable via google.

solomani, 2022-10-21:
Thanks Scires.