Atmosphere-NX/Atmosphere #1975: USB connection no longer working?

H4X0R46, 2022-10-22:
## Bug Report

[ If any section does not apply, replace its contents with "N/A". ]</br>
[ Lines between [ ] (square brackets) should be removed before posting. ]</br>

[ * ]</br>
[ Note: If the bug or crash you encountered is related to; ]</br>
[ - software used to make "backups", ]</br>
[ - software explicitly distributed for piracy, etc ]</br>
[ then contributors will not provide support for your issue and your issue will be closed. ]</br>

### What's the issue you encountered?
My switch no longer connects to my PC via USB connection. Doesn't even make the sound when plugged in and doesn't show ANYTHING different in device manager? Tried 2 different computers and two different USB C cables. No luck!
[ Describe the issue in detail and what you were doing beforehand. ]</br>
I just now updated my atmosphere to the official 1.4.0 build from the pre relase before it. 
[ Did you make any changes related to Atmosphère itself? ]</br>
Not really. I tried both with USB 3.0 forced and without in the system_settings.ini with no luck on either.
[ If so, make sure to include details relating to what exactly you changed. ]</br>

### How can the issue be reproduced?

[ * ]</br>
[ Include a detailed step by step process for recreating your issue. ]</br>
Just boot with newest atmosphere and try to do any USB related stuff with it. MTP transfer or anything, it won't show up in Windows at all.
### Crash Report

[ Crash reports can be found under ``/atmosphere/crash_reports``. ]</br>
[ If your issue caused Atmosphère to crash, include the crash report(s) by creating a [gist](https://gist.github.com/) and pasting the link here. ]</br>
No crash, console is running.
[ If you don't include a crash report in instances of crash related issues, we will ask you one to provide one. ]</br>

### System Firmware Version

15.0.0</br>
[ Replace X's with system firmware version at time of crash. ]</br>
[ You can find your firmware version in the Settings -> System, under "System Update". ]</br>
[ If it says "Update Pending", you can clear the pending update by rebooting to Maintenance Mode. ]</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: Hekate, newest build.
- Official release or unofficial build: Official
  - [ Official release version x.x.x (or) unofficial build ]
  - [ If using an unofficial build, include details on where/how you acquired the build. ] 
     - [ Ex: Self-compilation ]
     - [ Ex: Kosmos' distribution of Atmosphère ]
- Do you have additional kips or sysmodules you're loading: Just a nightly build of ldn_mitm.
- Homebrew software installed: [ * ]
- EmuMMC or SysNAND: EmuMMC and it's file based. FAT32
  - [ If using an EmuMMC, include whether it's partition-based or file-based. ]

### Additional context?

- Additional info about your environment:
- [ Any other information relevant to your issue. ]

SciresM, 2022-10-22:
Pretty sure this has nothing to do with atmospher.me. USB working totally fine over here.

H4X0R46, 2022-10-22:
What's weird is that it works fine inside hekate. with MTP, but when booted has this weird issue. Hmm.....

masagrator, 2022-10-22:
Can confirm on my Switch that with 1.4.0 / 15.0.0 MTP works fine

SciresM, 2022-10-22:
Does USB work when booted into stock? What custom sysmodules do you have? It would *really* surprise me if this was an ams thing, since ams doesn't touch usb at all except to enable usb3 when that setting is set.

H4X0R46, 2022-10-22:
I took a fresh sd card and made a new setup, creating a new EmuMMC now. I'll report back. I might just have corruption on the other.

H4X0R46, 2022-10-22:
The fresh setup works... Yeah looks like I'm gonna have to re do my main sd card 🤦‍♂️ That's annoying. Go ahead and close this issue. Oof!

closed 2022-10-22 by SciresM