Atmosphere-NX/Atmosphere #1976: Only 1.4.0 Error. A fatal error occurred when running Atmosphere.

nunssi, 2022-10-22:
hello!

I used Atmosphere 1.3.2 well this time, but to update to the new 1.4.0, I put the file and run it as usual.

=========================================

A fatal error occurred when running Atmosphere.
Program ID: 010041544d530000
Error Desc: std::abort() called (0xffe)

Failed to save report to the SD card! (0007d202)

Press POWER to reboot.

=========================================

I can't get it to work when it comes out like this.

So I thought it was damaged with chkdsk, so I tried to repair it, and even if I try to get the file back and put it back, that error keeps popping up.

It is the same even if I try Fix Archive Bit in the Hecate menu. I searched on the Internet, and there is no log even if I go to the log folder because the error log is saved, and I can't find it even when I look for the title number in the error message.

But the strange thing is that if I change it back to 1.3.2, it runs fine.

I didn't install a theme, and there's nothing wrong with the game icon, so why?

Even if I delete the required files and put them back in, 1.4.0 doesn't work, only 1.3.2 works, so why?

I searched other articles, found the system_settings file and commented it out telling me to comment ower_menu_reboot_function = str!payload , but the result is the same.

Masamune3210, 2022-10-22:
Please go back and actually fill out the issue template

SciresM, 2022-10-24:
@Muzika youre not getting the same issue, almost certainly. Open a new issue with your report file.

speaking of, @nunssi you didn't include your report file. Re-open with the report file.

SciresM, 2022-10-24:
(Failure to save report indicates SD card hardware problems, by the by)

SciresM, 2022-10-24:
Also, yeah, issue template not filled out. Locking this to make sure any further stuff here is in a new issue.

closed 2022-10-24 by SciresM