Atmosphere-NX/Atmosphere #1980: cheat toggles file ineffective without new line ending

zemixe, 2022-10-28:
## Bug Report

### What's the issue you encountered?

A minor issue concerning the reading of a cheat toggles file - assuming `dmnt_cheats_enabled_by_default = u8!0x0` is in use and instead a `toggles.txt` file is used to trigger selected cheats at game boot, the `toggles.txt` file will not be read unless ending in a new line, this is applicable to both CRLF and LF and this is not an issue with the main cheat file.

### How can the issue be reproduced?

1. Set `dmnt_cheats_enabled_by_default = u8!0x0` in `system_settings.ini`
2. Create a cheat file for any game
3. Create a toggles file next to the cheat file and activate any cheat at boot as follows:
```
[Cheat 1]
true
```
The cheat will not be toggled at boot.

4. Edit the toggles file as such:
```
[Cheat 1]
true
//newline
```
The cheat will now be toggled at boot.

Furthermore, the toggles feature is not documented at: 

https://github.com/Atmosphere-NX/Atmosphere/blob/master/docs/features/cheats.md

### System Firmware Version

15.0.0</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: hekate
- Atmosphere 1.4.0 official release
- Do you have additional kips or sysmodules you're loading: no
- Homebrew software installed: n/a
- SysNAND

SciresM, 2022-10-28:
The toggles feature is not actually intended to be edited by humans -- it is meant to be written by cheat management homebrew software like Edizon (and was designed largely at the request of the author of Edizon).

The lack of documentation is because I'm undecided on whether the feature is valuable -- the cheat engine itself will make its way into dmnt.gen2 for sure, not sure about the toggles.

Anyway, the relevant code is [here](https://github.com/Atmosphere-NX/Atmosphere/blob/master/stratosphere/dmnt/source/cheat/impl/dmnt_cheat_api.cpp#L1030), you're right that a string termination will result in failure-to-parse, though I think literally any whitespace character will work, not just a newline.

I guess this could potentially be fixed?

zemixe, 2022-10-28:
Thank you for the context. Usually I'll set things up manually where possible, which is how I came to encounter this. 

As regards whether or not the feature is valuable - personally I've found toggles to be very useful in separating "set and forget" cheats such as a graphics configuration (increased FPS, increased/decreased resolution) and "toggle when needed" situational cheats.

SciresM, 2022-10-28:
End-of-string should be equivalent to whitespace at the end of a toggle, now.

commit ff7a80e5 (2022-10-28): dmnt: allow string termination to end a toggle (closes #1980)
closed 2022-10-28 by SciresM